taks/esp32-nimble #4: Multiple connection to the server

nicolasfara, 2023-01-18:
Hi,

I need to connect multiple clients to the BLE server. Using your example code for the server, only one client at a time is connected. Is there a way to enable a multi-client connection?

Thank you!

taks, 2023-01-24:
By starting advertising when a connection is established, it is possible to respond to scans even while connected.
You can do it with the code below.

```rust
  let ble_device = BLEDevice::take();
  let server = ble_device.get_server();
  server.on_connect(|_| {
    ble_device.get_advertising().start();
  });
```

taks, 2023-01-24:
By default the maximum number of connections is 3.
If you want to change the maximum number of connections, please set `CONFIG_BT_NIMBLE_MAX_CONNECTIONS` in `sdkconfig.defaults`.

nicolasfara, 2023-01-27:
Ohh, thank you, @taks!! I'll try it out this weekend!

closed 2023-01-27 by nicolasfara