taks/esp32-nimble #14: Bonding issue

sploders101, 2023-04-20:
I'm not terribly familiar with BTLE pairing processes, so please forgive my ignorance, but I'm not able to get a connection to persist across resets. It looks like the process I need is called "bonding", which I have enabled (it is also enabled in the keyboard example, which is what I'm primarily interested in).

I have tried a significant number of things, going so far as manually adding `extern` blocks to pull in functionality from ESP-IDF that the crate doesn't expose. I have tried enabling RPD by patching the library, using `CONFIG_BT_NIMBLE_NVS_PERSIST=y` in my `sdkconfig.defaults` file, and much more. It looked like I almost got it with RPD enabled, but I got some error about a resolve list being missing and couldn't find any information on the topic.

I'm hoping someone else who has had this issue can lead me in the right direction? How can I get a bonded connection to persist across resets?

taks, 2023-04-24:
I tested on Windows 10 and [examples/ble_secure_server.rs](https://github.com/taks/esp32-nimble/blob/develop/examples/ble_secure_server.rs) code.
I rebooted both devices and was able to maintain the connection.
Please let me know the devices and the program you tested.

Skizy, 2023-07-17:
Hey, I'm also experiencing this issue. I can't keep connection after the device reset. I have tested it both on my Linux machine and my Android phone. I am using the [examples/ble_keyboard.rs](https://github.com/taks/esp32-nimble/blob/develop/examples/ble_keyboard.rs) code. My esp is esp32-s3.
On my Linux machine after resetting the device it start to connect and disconnect multiple times until I stop it. On my phone it connects and immediately disconnects, but only one time and stays disconnected.

taks, 2023-07-19:
Thanks for the report.
I reproduced it in my environment.

I checked sample program and noticed that the call to the initialization function (`ble_store_config_init`) is missing.
I added a call to this function and confirmed that it works after resetting.

closed 2023-07-19 by taks