taks/esp32-nimble #17: Sending Characteristic between threads

Skizy, 2023-05-17:
Hey, I was trying to create a separate thread to send notifications as a server, but was unable to, because BLECharacteristic is not Send and not Sync. Hence I have a question: what is the point of wrapping characteristics in Arc<Mutex> if you can't share them between threads?

taks, 2023-05-18:
Please use Arc::clone.
See https://doc.rust-lang.org/book/ch16-03-shared-state.html#atomic-reference-counting-with-arct for details.
If it still doesn't work, please paste the source code.

Skizy, 2023-05-19:
Something like this for example does not compile. BLECharacteristic doesn't implement Sync and Send because Vec<ble_gatt_sdc_def> does not, since ble_gatt_sdc_def contains *mut c_void.

```rust
use esp32_nimble::*;
use std::sync::Arc;

let ble_dev = BLEDevice::take();

let server = ble_dev.get_server();

let service = server.create_service(utilities::BleUuid::from_uuid16(9234u16));
let notify_characteristic = service.lock().create_characteristic(
    utilities::BleUuid::from_uuid16(1642u16),
    NimbleProperties::NOTIFY | NimbleProperties::READ,
);
notify_characteristic.lock().set_value(&[125]);

let char_clone = Arc::clone(&notify_characteristic);
let service_clone = Arc::clone(&service);

std::thread::spawn(move || {
    char_clone.lock().notify();
});
```

Skizy, 2023-05-19:
Thanks a lot! You should also mark `BLEService` as `Send`, as it has the same problem.

commit 9e7d776e (2023-05-19): mark BLECharacteristic as Send (Fix #17)
closed 2023-05-19 by Skizy