taks/esp32-nimble #20: Bonding breaks secure connection

Skizy, 2023-07-05:
Hey, I am trying to use bonding and secure connection on 2 esp32-s3's.
Using your examples of secure server and secure client everything works the first time. But after I reload the client, they connect, but the client hangs on secure connection, and the server prints, that an unhandled event happens (esp32_nimble::server::ble_server: unhandled event: 17) (If i checked correctly, 17 is BLE_GAP_EVENT_REPEAT_PAIRING). After that client just times out.

Without bonding everything works just fine.

I used your examples without changing them, so I don't think I have to post the code.

I am using esp-idf v4.4 with the help of esp-idf-template, if that's in any way helpful.

taks, 2023-07-07:
Thanks for the report.
I fixed the bug.

You can check the operation by adding the following to your Cargo.toml.
```
[patch.crates-io]
esp32-nimble = { git = 'https://github.com/taks/esp32-nimble.git', rev = "63015eec2cef5575abf491f1e3795ffe69f6cc04" }
```

Skizy, 2023-07-07:
Yes, everything works as expected now, thank you!

closed 2023-07-07 by Skizy