taks/esp32-nimble #21: ESP32C3 BLE advertising not work

zgdump, 2023-08-02:
Hello!

Thanks for the library. But I am unable to use it( I try to run the example `ble_server.rs` and this example starts advertising, my phone sees the ESP32, but when I try to pair it, nothing happens. Literally nothing: the phone does not respond and there is silence in the ESP32 logs.

What am I doing wrong?

<details>
  <summary>Logs</summary>
  
  ```
  ESP-ROM:esp32c3-api1-20210207
  Build:Feb  7 2021
  rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
  SPIWP:0xee
  mode:DIO, clock div:2
  load:0x3fcd5820,len:0x171c
  0x3fcd5820 - _bss_end
      at ??:??
  load:0x403cc710,len:0x968
  0x403cc710 - _iram_bss_start
      at ??:??
  load:0x403ce710,len:0x2f68
  0x403ce710 - _iram_bss_start
      at ??:??
  SHA-256 comparison failed:
  Calculated: 1d06b938c0222bf626e0bdf46178b1b37ab24d03f0360fc8fcf7153c2571deaf
  Expected: 68d7bdf643ba446b8ed7ae8423241d442fd052b2bc77091100ba06fd65dcf8d5
  Attempting to boot anyway...
  entry 0x403cc710
  0x403cc710 - _iram_bss_start
      at ??:??
  I (50) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
  I (50) boot: compile time Jun  7 2023 07:59:10
  I (51) boot: chip revision: v0.3
  I (55) boot.esp32c3: SPI Speed      : 40MHz
  I (60) boot.esp32c3: SPI Mode       : DIO
  I (65) boot.esp32c3: SPI Flash Size : 4MB
  I (69) boot: Enabling RNG early entropy source...
  I (75) boot: Partition Table:
  I (78) boot: ## Label            Usage          Type ST Offset   Length
  I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
  I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
  I (101) boot:  2 factory          factory app      00 00 00010000 003f0000
  I (108) boot: End of partition table
  I (112) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=301e8h (197096) map
  I (164) esp_image: segment 1: paddr=00040210 vaddr=3fc91200 size=02224h (  8740) load
  I (166) esp_image: segment 2: paddr=0004243c vaddr=40380000 size=0dbdch ( 56284) load
  I (184) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=86824h (550948) map
  I (305) esp_image: segment 4: paddr=000d684c vaddr=4038dbdc size=03598h ( 13720) load
  I (313) boot: Loaded app from partition at offset 0x10000
  I (314) boot: Disabling RNG early entropy source...
  I (325) cpu_start: Unicore app
  I (325) cpu_start: Pro cpu up.
  I (334) cpu_start: Pro cpu start user code
  I (334) cpu_start: cpu freq: 160000000 Hz
  I (334) cpu_start: Application information:
  I (337) cpu_start: Project name:     libespidf
  I (342) cpu_start: App version:      1
  I (347) cpu_start: Compile time:     Aug  2 2023 22:15:12
  I (353) cpu_start: ELF file SHA256:  0000000000000000...
  I (359) cpu_start: ESP-IDF:          3187b8b
  I (364) cpu_start: Min chip rev:     v0.3
  I (368) cpu_start: Max chip rev:     v0.99 
  I (373) cpu_start: Chip rev:         v0.3
  I (378) heap_init: Initializing. RAM available for dynamic allocation:
  I (385) heap_init: At 3FC95E70 len 000468A0 (282 KiB): DRAM
  I (391) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
  I (398) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
  I (406) spi_flash: detected chip: generic
  I (409) spi_flash: flash io: dio
  W (413) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
  W (424) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
  I (432) sleep: Configure to isolate all GPIO pins in sleep state
  I (439) sleep: Enable automatic switching of GPIO sleep configuration
  I (446) coexist: coex firmware version: 80b0d89
  I (452) coexist: coexist rom version 9387209
  I (457) app_start: Starting scheduler on CPU0
  I (461) main_task: Started on CPU0
  I (461) main_task: Calling app_main()
  I (471) BLE_INIT: BT controller compile version [963cad4]
  I (471) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
  W (481) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
  I (531) BLE_INIT: Bluetooth MAC: 34:b4:72:4e:1a:ea
  
  I (541) esp32_nimble::ble_device: BLE Host Task Started
  I (541) NimBLE: GAP procedure initiated: stop advertising.
  
  I (541) esp32_nimble::ble_device: Device Address: 34:B4:72:4E:1A:EA
  primary service
             uuid 0x1800
           handle 1
       end_handle 5
  characteristic
             uuid 0x2a00
       def_handle 2
       val_handle 3
     min_key_size 0
            flags [READ]
  characteristic
             uuid 0x2a01
       def_handle 4
       val_handle 5
     min_key_size 0
            flags [READ]
  primary service
             uuid 0x1801
           handle 6
       end_handle 9
  characteristic
             uuid 0x2a05
       def_handle 7
       val_handle 8
     min_key_size 0
            flags [INDICATE]
  ccc descriptor
             uuid 0x2902
           handle 9
     min_key_size 0
            flags [READ|WRITE]
  primary service
             uuid fafafafa-fafa-fafa-fafa-fafafafafafa
           handle 10
       end_handle 17
  characteristic
             uuid d4e0e0d0-1a2b-11e9-ab14-d663bd873d93
       def_handle 11
       val_handle 12
     min_key_size 0
            flags [READ]
  characteristic
             uuid a3c87500-8ed3-4bdf-8a39-a01bebede295
       def_handle 13
       val_handle 14
     min_key_size 0
            flags [READ|NOTIFY]
  ccc descriptor
             uuid 0x2902
           handle 15
     min_key_size 0
            flags [READ|WRITE]
  characteristic
             uuid 3c9a3f00-8ed3-4bdf-8a39-a01bebede295
       def_handle 16
       val_handle 17
     min_key_size 0
            flags [READ|WRITE]
  I (661) NimBLE: GAP procedure initiated: advertise; 
  I (671) NimBLE: disc_mode=2
  I (671) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
  I (681) NimBLE: 
  ```
</details>

<details>
  <summary>Source code</summary>

  ```rust
  #![no_std]
  #![no_main]
  
  extern crate alloc;
  
  use alloc::format;
  use esp32_nimble::{uuid128, BLEDevice, NimbleProperties};
  use esp_idf_sys as _;
  
  #[no_mangle]
  fn main() {
      esp_idf_sys::link_patches();
      esp_idf_svc::log::EspLogger::initialize_default();
  
      let ble_device = BLEDevice::take();
  
      let server = ble_device.get_server();
      server.on_connect(|_| {
          ::log::info!("Client connected");
          ::log::info!("Multi-connect support: start advertising");
          ble_device.get_advertising().start().unwrap();
      });
      let service = server.create_service(uuid128!("fafafafa-fafa-fafa-fafa-fafafafafafa"));
  
      // A static characteristic.
      let static_characteristic = service.lock().create_characteristic(
          uuid128!("d4e0e0d0-1a2b-11e9-ab14-d663bd873d93"),
          NimbleProperties::READ,
      );
      static_characteristic
          .lock()
          .set_value("Hello, world!".as_bytes());
  
      // A characteristic that notifies every second.
      let notifying_characteristic = service.lock().create_characteristic(
          uuid128!("a3c87500-8ed3-4bdf-8a39-a01bebede295"),
          NimbleProperties::READ | NimbleProperties::NOTIFY,
      );
      notifying_characteristic.lock().set_value(b"Initial value.");
  
      // A writable characteristic.
      let writable_characteristic = service.lock().create_characteristic(
          uuid128!("3c9a3f00-8ed3-4bdf-8a39-a01bebede295"),
          NimbleProperties::READ | NimbleProperties::WRITE,
      );
      writable_characteristic
          .lock()
          .on_read(move |_, _| {
              ::log::info!("Read from writable characteristic.");
          })
          .on_write(move |value, _param| {
              ::log::info!("Wrote to writable characteristic: {:?}", value);
          });
  
      let ble_advertising = ble_device.get_advertising();
      ble_advertising
          .name("ESP32-GATT-Server")
          .add_service_uuid(uuid128!("fafafafa-fafa-fafa-fafa-fafafafafafa"));
  
      ble_advertising.start().unwrap();
  
      let mut counter = 0;
      loop {
          log::info!("Pun!");
  
          esp_idf_hal::delay::FreeRtos::delay_ms(1000);
  
          notifying_characteristic
              .lock()
              .set_value(format!("Counter: {counter}").as_bytes())
              .notify();
  
          counter += 1;
      }
  }
  ```

</details>

<details>
  <summary>sdkconfig.default</summary>
  
  ```
  # Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
  CONFIG_ESP_MAIN_TASK_STACK_SIZE=7000
  
  CONFIG_BT_ENABLED=y
  CONFIG_BT_BLE_ENABLED=y
  CONFIG_BT_BLUEDROID_ENABLED=n
  CONFIG_BT_NIMBLE_ENABLED=y
  ```
</details>

Rust version: `1.73.0-nightly (7ac9416d8 2023-07-24)`
Rust target: `riscv32imc-esp-espidf`
Rust flags: `rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]`
Idf version: `release/v5.1`

taks, 2023-08-07:
The example `ble_server.rs` is a code that does not use encryption.
In this case, pairing is not required and communication is possible.

If you want to try the code using pairing (encryption), please refer to `examples/ble_secure_server.rs`.

zgdump, 2023-08-12:
I didn't know that BLE doesn't always require pairing! Previously, I only programmed Bluetooth Classic. Thank you very much for correcting.

I try security server example and it works! It is strange that I did not notice it earlier and did not try it. Thanks again for the good library.

closed 2023-08-12 by zgdump