taks/esp32-nimble #22: [Feature request]: Set maximum lenght of a characteristic

oyren, 2023-08-06:
NimBLE-Arduino has the possibility to set a maximum length of bytes which a characteristic supports (https://h2zero.github.io/NimBLE-Arduino/class_nim_b_l_e_characteristic.html#aa32224e69115e9ba60ac614e8f37a871`max_length`), i don't see this possibility here.

In my opinion it would be useful to reject some kind of values.
Or is there an option to reject values e.g. in the `on_write` callback which i have overlooked?

Thanks in advance.

WDYT?

taks, 2023-08-09:
Added the feature to reject received data in the on_write callback.
Please check.

systec-ms, 2023-08-09:
Thank you, i appreciate your work.
The change to simply don't set the new value (if the reject flag is set to `true`) is already quite useful for me and works as expected.

Disclaimer: Here's what I wanted to note/suggest, but as it turns out, I don't understand the spec [1] (maybe you have more knowledge in this field).

However, an even better solution (IMHO) would be to report an error code like specified in the BLE spec (if possible).

For example, i could report the error code `0x0105` (Insufficient authentication) or `0x010D` Invalid value size.
But as mentioned above, it seems like i have some missing information here, since i don't see the bridge between the spec error codes ([1] page 370)  and the codes listed
[here](https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v4.x.x/doc/html/group___b_l_e___g_a_t_t___s_t_a_t_u_s___c_o_d_e_s.html).

[1] BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 1, Part F  (Revision Date: 2021-07-13)

systec-ms, 2023-08-09:
Thanks, the link was very helpful.