taks/esp32-nimble #23: Latency and connection interval

nmarks99, 2023-08-07:
I am wondering if the latency I am experiencing is normal and or if there is a way to speed it up. With a simple notifying characteristic running in a loop with a 1ms delay updating the value of the characteristic (integer that increase by one each time), the read operation on the client side takes approximately 90-100ms which to me seems much too slow. In other libraries such as the Arduino NimBLE library, they seem to have an option for "connection interval" which essentially allows you to increase transmission rates at the cost of power consumption. Is there an option for this in this library?

taks, 2023-08-08:
The set_connection_params and update_conn_params functions added.
Please check.

nmarks99, 2023-08-09:
Works now, thanks for the quick update!

closed 2023-08-09 by nmarks99