taks/esp32-nimble #24: Errors with iPhone pairing

LarryMerino, 2023-08-16:
I have the following blE server
```rust
let ble_device = Esp32BleDevice::take();
        ble_device
            .security()
            .set_auth(true, true, true)
            .set_passkey(ble_configuration.pin_number)
            .set_io_cap(SecurityIOCap::DisplayOnly);
```

And the following characteristic 
```rust
let settings_characteristic = main_service.lock().create_characteristic(
            BleUuid::from_uuid128_string(SETTINGS_CHARACTERISTIC_UUID).unwrap(),
            NimbleProperties::READ
                | NimbleProperties::READ_ENC
                | NimbleProperties::READ_AUTHEN
                | NimbleProperties::WRITE
                | NimbleProperties::WRITE_ENC
                | NimbleProperties::WRITE_AUTHEN
        );
```
When I try to read there from an android, there is no problem. 
The problem comes with iPhone connections.
If I use NimbleProperties::READ_AUTHEN or NimbleProperties::WRITE_AUTHEN I can connect to the esp32-c3 but I can't read or write on that characteristic. It always says `Error Reading Characteristic 68F... Authentication is insufficient`
If I remove NimbleProperties::READ_AUTHEN or NimbleProperties::WRITE_AUTHEN
```rust
let settings_characteristic = main_service.lock().create_characteristic(
            BleUuid::from_uuid128_string(SETTINGS_CHARACTERISTIC_UUID).unwrap(),
            NimbleProperties::READ
                | NimbleProperties::READ_ENC
                | NimbleProperties::WRITE
                | NimbleProperties::WRITE_ENC
        );
```
I can use that characteristic but the first time I try to read from there It says: `Error Reading Characteristic 68F... Authentication is insufficient` but when I try to read again, it works.

The last thing that I'm not able to solve is when I try to disconnect and connect again from iPhone device. No matter if I use NimbleProperties::READ_ENC or NimbleProperties::READ_ENC and NimbleProperties::READ_AUTHEN I'm not able to reconnect that device again from iPhone. It says `Failed to Connect: Peer removed pairing information`

It looks like esp32-c3 is not able to save some kind of data that iPhone needs to connect with this device again.

I make some research and seems to have something to do with LTK (Long Term Key) and RPA (Resolvable Private Address) but I'm not able to solve it.

Does anyone have or have had the same problem?

This is my sdkconfig.defaults file 
```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=7000

# Use this to set FreeRTOS kernel tick frequency to 1000 Hz (100 Hz by default).
# This allows to use 1 ms granuality for thread sleeps (10 ms by default).
#CONFIG_FREERTOS_HZ=1000

# Workaround for https://github.com/espressif/esp-idf/issues/7631
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=n

# Bluetooth
CONFIG_BT_ENABLED=y
CONFIG_BT_BLE_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_NVS_PERSIST=y

CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
```

This is my partition table
```
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs,      data, nvs,      0x9000,  0x4000,
otadata,  data, ota,      0xd000,  0x2000,
phy_init, data, phy,      0xf000,  0x1000,
factory,  app,  factory,  0x10000,  1M,
ota_0,    app,  ota_0,    ,         1M,
ota_1,    app,  ota_1,    ,         1M,
report_data,  data, nvs, ,        0x12000
```
🤔 I am doing something wrong?

LarryMerino, 2023-08-17:
Thanks @taks for your interest

* esp32-nimble version 0.2.0
* esp-idf version 5.1 (I try to use 4.4.5 to follow your configuration but I'm not able to compile it. Maybe that is the problem? 🤔)

I try to reproduce the code using ESP-IDF version 4.4.5 but there seems to be some problems compiling version 4.X of the esp-idf [#149](https://github.com/esp-rs/esp-idf-template/issues/149)

* I see a warning, that's all `W (126786) esp32_nimble::server::ble_server: unhandled event: 18`
```
I (126386) fw::devices::ble_device::ble_device: BLEDevice:  on_connect event launched
I (126386) fw::devices::ble_device::ble_device: BLEDevice:  Connection with: [82, 206, 232, 201, 237, 112]
I (126396) NimBLE: GAP procedure initiated: advertise;
I (126396) NimBLE: disc_mode=2
I (126406) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (126416) NimBLE:

W (126786) esp32_nimble::server::ble_server: unhandled event: 18
I (126876) esp32_nimble::server::ble_server: mtu update event; conn_handle=1 mtu=256
I (146376) fw::devices::ble_device::ble_device: BLEDevice:  on_disconnect event launched
I (146376) fw::devices::ble_device::ble_device: BLEDevice:  Disconnection with: [82, 206, 232, 201, 237, 112]
```

However when I connect from an android the logs are a bit different
```
I (107456) fw::devices::ble_device::ble_device: BLEDevice:  on_connect event launched
I (107456) fw::devices::ble_device::ble_device: BLEDevice:  Connection with: [83, 6, 160, 46, 183, 220]
I (107466) NimBLE: GAP procedure initiated: advertise;
I (107476) NimBLE: disc_mode=2
I (107476) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (107486) NimBLE:

I (121606) esp32_nimble::server::ble_server: AuthenticationComplete
W (121606) esp32_nimble::server::ble_server: unhandled event: 16
I (121616) NimBLE: GAP procedure initiated: stop advertising.

W (121616) esp32_nimble::server::ble_server: unhandled event: 9
```
* Regarding Arduino, I know it exists but I never use it because I don't know C or C++ and I'm not familiar with this platform 😔

taks, 2023-08-18:
Thanks for your reply.
- Regarding the warning, the 18 code is `BLE_GAP_EVENT_PHY_UPDATE_COMPLETE`.
  I don't think it is relevant in this issue.
- I have created [a sample program](https://github.com/taks/NimBLE-Arduino-Test) using the c++ library. 
  I am not sure if this will be helpful since the version is different from the one used in this case, 
  but is it possible to try?
  (Instructions on how to try it are included in the project's README.)

LarryMerino, 2023-09-07:
> Thanks for your reply.
>
> * Regarding the warning, the 18 code is `BLE_GAP_EVENT_PHY_UPDATE_COMPLETE`.
>   I don't think it is relevant in this issue.
> * I have created [a sample [...]

I have tried with the platformIO example that you provided and the behavior is the same so I assume that the problem has to do with the way the BLE interacts with the iPhone rather than with the library itself.
Thanks for the attention.

The logs using platformIO example
```
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onConnect(): Default
D NimBLEServerCallbacks: onConnect(): Default
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGATTServerEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: mtu update event; conn_handle=1 mtu=255
D NimBLEServerCallbacks: onMTUChange(): Default
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onPassKeyRequest: default: 123456
D NimBLEServer: BLE_SM_IOACT_DISP; ble_sm_inject_io result: 0
D NimBLEServer: << handleGATTServerEvent
D NimBLEServer: >> handleGapEvent: 
I NimBLEServer: subscribe event; attr_handle=8, subscribed: false
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onDisconnect(): Default
D NimBLEServerCallbacks: onDisconnect(): Default
D NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0
D NimBLEAdvertising: << Advertising start
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onConnect(): Default
D NimBLEServerCallbacks: onConnect(): Default
D NimBLEServer: >> handleGapEvent: 
D NimBLEServer: << handleGATTServerEvent
D NimBLEServer: >> handleGapEvent: 
D NimBLEServerCallbacks: onDisconnect(): Default
D NimBLEServerCallbacks: onDisconnect(): Default
D NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0
D NimBLEAdvertising: << Advertising start
```

LarryMerino, 2023-11-07:
Hi @taks 👋🏽
It seems that iOS uses "Resolvable Private Address (RPA)" for Bluetooth so to solve this issue, I followed the instructions I found in this [post]( https://www.esp32.com/viewtopic.php?t=14966#p58984) and now works perfect.
In order to follow the instructions I have this:
```rust
unsafe {
    esp_idf_sys::ble_hs_cfg.sm_our_key_dist =
        (BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID) as u8;
}
unsafe {
    esp_idf_sys::ble_hs_cfg.sm_their_key_dist =
        (BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID) as u8;
}
```
I think it could be interesting to include this configuration within the library. Something like this:
```rust
        let ble_device = BLEDevice::take();
        ble_device
            .security()
            .set_auth(true, true, true)
            .set_passkey(ble_configuration.pin_number)
            .set_io_cap(SecurityIOCap::DisplayOnly);
            .resolve_rpa(true)
```
but it's just a suggestion

I close this issue but if you have any questions that I can help you with or if you want to implement RPA as a new feature, feel free to reopen the issue

Thank you very much again

closed 2023-11-07 by LarryMerino

taks, 2023-11-08:
Thanks @LarryMerino.

Can I ask you a question?

In the post, `ble_hs_pvcy_rpa_config` function was mentioned.
Did you also add this function call when you got it working?

LarryMerino, 2023-11-14:
Can work without this parameter. This parameter is for the iPhone to save the Bluetooth device to the system.
We have commented on it
```rust
// This thing below makes iOS device NOT FORGET THE BLE device!
// unsafe { esp_idf_sys::ble_hs_pvcy_rpa_config(NIMBLE_HOST_ENABLE_RPA as u8); }
```

It depends on the use case, in our case it is neither necessary nor desirable for the iPhone to save the BLE device in the system

PR #39 (2023-11-14): Add resolve_rpa function
Fix #24