taks/esp32-nimble #37: How to call esp_ble_gap_set_rand_addr?

lz1998, 2023-11-12:
It's in [BLEAdvertising.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEAdvertising.cpp#L286)

taks, 2023-11-14:
I added `BLEDevice.set_own_addr_type` and `BLEDevice.set_rnd_addr` functions.
You can use the following

```lang=rust
let ble_device = BLEDevice::take();
ble_device.set_own_addr_type(OwnAddrType::Random);
let addr = [0xFCu8, 0xA2u8, 0xD5u8, 0x14u8, 0x5Du8, 0x98u8];
ble_device.set_rnd_addr(addr).unwrap();
```

lz1998, 2023-11-15:
```bash
  = note: [ldproxy] Running ldproxy
          Error: Linker /Users/lz1998/RustroverProjects/evil-apple-juice-esp32-rs/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /Users/lz1998/RustroverProjects/evil-apple-juice-esp32-rs/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/lz1998/RustroverProjects/evil-apple-juice-esp32-rs/target/riscv32imc-esp-espidf/debug/deps/libesp32_nimble-8f23e8b04dddce3f.rlib(esp32_nimble-8f23e8b04dddce3f.esp32_nimble.671d7f6fcb67d447-cgu.05.rcgu.o): in function `esp32_nimble::ble_device::BLEDevice::_set_own_addr_type':
          /Users/lz1998/.cargo/git/checkouts/esp32-nimble-fc9d2c54a3bfa7fe/8250ca8/src/ble_device.rs:(.text._ZN12esp32_nimble10ble_device9BLEDevice18_set_own_addr_type17h1ce402db9c242dddE+0x3c): undefined reference to `ble_hs_pvcy_rpa_config'
          collect2: error: ld returned 1 exit status
          
          
          Stack backtrace:
             0: std::backtrace_rs::backtrace::libunwind::trace
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5
             1: std::backtrace_rs::backtrace::trace_unsynchronized
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
             2: std::backtrace::Backtrace::create
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/backtrace.rs:331:13
             3: std::backtrace::Backtrace::capture
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/backtrace.rs:297:9
             4: anyhow::error::<impl anyhow::Error>::msg
             5: ldproxy::main
             6: std::sys_common::backtrace::__rust_begin_short_backtrace
             7: std::rt::lang_start::{{closure}}
             8: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/core/src/ops/function.rs:284:13
             9: std::panicking::try::do_call
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/panicking.rs:552:40
            10: std::panicking::try
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/panicking.rs:516:19
            11: std::panic::catch_unwind
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/panic.rs:142:14
            12: std::rt::lang_start_internal::{{closure}}
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/rt.rs:148:48
            13: std::panicking::try::do_call
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/panicking.rs:552:40
            14: std::panicking::try
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/panicking.rs:516:19
            15: std::panic::catch_unwind
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/panic.rs:142:14
            16: std::rt::lang_start_internal
                       at /rustc/dd430bc8c22f57992ec1457a87437d14283fdd65/library/std/src/rt.rs:148:20
            17: std::rt::lang_start
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
```

lz1998, 2023-11-15:
```toml
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]

[unstable]
build-std = ["core", "alloc", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"
```

closed 2023-11-15 by taks

taks, 2023-11-15:
Fixed in #44.

PR #40 (2023-11-14): Added functions to set own address.
Fix #37.
PR #44 (2023-11-15): Fix link error for esp32c3
Fix #37 link error

reference: https://github.com/espressif/esp-idf/issues/11893#issuecomment-1639549702