taks/esp32-nimble #38: How to set adv_type?

lz1998, 2023-11-12:
https://docs.espressif.com/projects/esp-idf/zh_CN/release-v5.1/esp32/api-reference/bluetooth/esp_gap_ble.html#_CPPv418esp_ble_adv_type_t

taks, 2023-11-14:
I added `BLEAdvertising.advertisement_type` function.
Please check.

lz1998, 2023-11-15:
> I added `BLEAdvertising.advertisement_type` function. Please check.

There are 5 modes in the enum.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_ble.html#_CPPv418esp_ble_adv_type_t

lz1998, 2023-11-16:
How to set the other two types?

taks, 2023-11-17:
I have researched the two other enum.

In nimble, it appears to be set up as follows
https://github.com/espressif/esp-nimble/blob/bdf2c7588a5c7d0c16e98a168b0cf41578b119ef/nimble/host/src/ble_gap.c#L2428-L2453

I will modify the program to allow setting the relevant parameters.

PR #49 (2023-11-18): Add `disc_mode` and `high_duty_cycle` functions.
Fix #38.
closed 2023-11-18 by taks