taks/esp32-nimble #42: cargo check error

zzn26, 2023-11-15:
When I add this dependency the check fails.
Please help me see what the problem is.
Cargo.toml
```
[package]
name = "esp32-test"
version = "0.1.0"
authors = ["zzn"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.47.3", default-features = false }
esp32-nimble = "0.3.0"

[build-dependencies]
embuild = "0.31.3"

```
cargo check result:
```
    Checking embedded-svc v0.26.4
error[E0053]: method `acquire` has an incompatible type for trait
   --> /home/zzn/.cargo/registry/src/rsproxy.cn-0dccff568467c15b/esp-idf-hal-0.42.5/src/task.rs:749:29
    |
749 |         unsafe fn acquire() {
    |                             ^ expected `bool`, found `()`
    |
    = note: expected signature `unsafe fn() -> bool`
               found signature `unsafe fn()`
help: change the output type to match the trait
    |
749 |         unsafe fn acquire() -> bool {
    |                             +++++++

error[E0053]: method `release` has an incompatible type for trait
   --> /home/zzn/.cargo/registry/src/rsproxy.cn-0dccff568467c15b/esp-idf-hal-0.42.5/src/task.rs:753:35
    |
753 |         unsafe fn release(_token: ()) {
    |                                   ^^
    |                                   |
    |                                   expected `bool`, found `()`
    |                                   help: change the parameter type to match the trait: `bool`
    |
    = note: expected signature `unsafe fn(bool)`
               found signature `unsafe fn(())`

For more information about this error, try `rustc --explain E0053`.
error: could not compile `esp-idf-hal` (lib) due to 2 previous errors
```
Thx

taks, 2023-11-15:
The cause for this error is that the `"esp-idf-svc/critical-section"` in your Cargo.toml conflicts with the `"critical-section/std"` used by esp32-nimble.

Remove your `"esp-idf-svc/critical-section"` or change `esp32-nimble = "0.3.0"` to `esp32-nimble = { version = "0.3.0", default-features = false }` and it should work.

※ I would like to review the esp32-nimble library to prevent this error in the future.

zzn26, 2023-11-16:
> The cause for this error is that the `"esp-idf-svc/critical-section"` in your Cargo.toml conflicts with the `"critical-section/std"` used by esp32-nimble.
>
> Remove your [...]

It worked. Thank you very much.

closed 2023-11-16 by zzn26