taks/esp32-nimble #46: Unresolved import `core::sync::atomic::AtomicU64`

DerFetzer, 2023-11-17:
I am trying to use this library with a `ESP32-C3` unfortunately when adding `esp32-nimble` to the dependencies I get a compiler error:
```
error[E0432]: unresolved import `core::sync::atomic::AtomicU64`
   --> /home/derfetzer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-polyfill-1.0.3/src/polyfill.rs:6:17
    |
6   |         pub use core::sync::atomic::$atomic_type;
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `AtomicU64` in `sync::atomic`
...
169 | atomic_int!(u64, AtomicU64, polyfill_u64);
    | -----------------------------------------
    | |                |
    | |                help: a similar name exists in the module: `AtomicU32`
    | in this macro invocation
    |
    = note: this error originates in the macro `atomic_int` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0432]: unresolved import `core::sync::atomic::AtomicI64`
   --> /home/derfetzer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-polyfill-1.0.3/src/polyfill.rs:6:17
    |
6   |         pub use core::sync::atomic::$atomic_type;
    |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `AtomicI64` in `sync::atomic`
...
174 | atomic_int!(i64, AtomicI64, polyfill_i64);
    | -----------------------------------------
    | |                |
    | |                help: a similar name exists in the module: `AtomicI32`
    | in this macro invocation
    |
    = note: this error originates in the macro `atomic_int` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0432`.
error: could not compile `atomic-polyfill` (lib) due to 2 previous errors

```

I am using the `esp-idf-template` and only added the `CONFIG_BT_*` flags and this library as a dependency.

It seems there have been issues with `Xtensa` targets in the past (https://github.com/embassy-rs/atomic-polyfill/issues/14).
The last hint there was to update the compiler version but I did `espup update` just today.

Do you have any idea what the problem could be?

Thanks in advance!

taks, 2023-11-18:
Please write the contents of your .cargo/config.toml and Cargo.toml, and the compiler version.

DerFetzer, 2023-11-18:
Sure, thanks for your quick reply!

`config.toml`
```toml
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"
```

`Cargo.toml`
```toml
[package]
name = "project-d"
version = "0.1.0"
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.47.3", default-features = false }
esp32-nimble = "0.3.1"

[build-dependencies]
embuild = "0.31.3"
```
Compiler
```bash
~/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/rustc  --version                                                                                                        
rustc 1.76.0-nightly (a57770440 2023-11-16)
```

taks, 2023-11-18:
I found that the critical-section feature in the once_cell library is causing this problem.
I would change to use the std feature in the once_cell library.

closed 2023-11-18 by taks

DerFetzer, 2023-11-19:
Thank you very much! Works like a charm!

PR #48 (2023-11-18): Use once_cell/std
Fix #46