taks/esp32-nimble #57: Reading connection strength

lissyx, 2023-12-12:
Not really an issue but rather a question, so if this is not the right place for that, I'd be happy to post where it should be.

I've been using `esp32-nimble` crate a for a small DIY project (basically doing https://www.thingiverse.com/thing:2146425 but with `esp32-c3`, Rust, and over Bluetooth). One thing I would like to add to my remote controller is connection strength information.

There's a `get_power` on `BLEDevice` https://taks.github.io/esp32-nimble/esp32_nimble/struct.BLEDevice.html#method.get_power but I suspect it just returns the **configured** power transmit, and not the actual signal level ?

There is also `rssi` exposed on `BLEAdvertisedDevice` https://taks.github.io/esp32-nimble/esp32_nimble/struct.BLEAdvertisedDevice.html#method.rssi ; does it means one need to keep track of the `BLEAdvertisedDevice` after connection to keep the signal information, or this is only valid after sanning for bluetooth devices ? (I suspect it is the latter).

Searching for RSSI on Espressif's docs yields only https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32c3/api-reference/bluetooth/esp_gap_ble.html?highlight=read_rssi#_CPPv421esp_ble_gap_read_rssi13esp_bd_addr_t which I suspect is what is exposed by `BLEAdverisedDevice::rssi()` ?

So, question is: is there a way to get signal strength on a live connection ? Or one of the above is the right solution ?

taks, 2023-12-13:
Question is welcome.

There is a function to obtain the RSSI of the last communication.
https://taks.github.io/esp32-nimble/esp_idf_sys/fn.ble_gap_conn_rssi.html

Currently, BLEClient does not have an API that uses this function.
I will add an API to my program that uses this function.

lissyx, 2023-12-13:
That was quick, thanks !

lissyx, 2023-12-14:
Tested and my two esp32-c3 boards being ~30cm apart, it reports -38~-41 of RSSI (I suspect it is reported in dBm)

taks, 2023-12-14:
`Received signal strength indication in dBm` is mentioned in https://github.com/espressif/esp-nimble/blob/bdf2c7588a5c7d0c16e98a168b0cf41578b119ef/nimble/host/include/host/ble_gap.h#L431

PR #58 (2023-12-13): Added BLEClient::get_rssi
Fix #57