taks/esp32-nimble #59: no_std not compiling

thedevleon, 2023-12-22:
I already have esp32-nimble working in a std context just fine, but I would also like to get it running in a no_std context as well, since there seems to be a no_std feature. 

However, I'm unable to get esp32-nimble to build, getting the following error:
```
error: failed to run custom build command for `esp-idf-sys v0.33.7`

Caused by:
  process didn't exit successfully: `/home/leon/Desktop/dev/esp32c6-no-std/target/debug/build/esp-idf-sys-823049c3c6f61c3a/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: None,
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: Unsupported target 'riscv32imac-unknown-none-elf'
```

Please see [this repo](https://github.com/thedevleon/esp32-nimble-no-std) for an example setup. I've also tried with the esp32c3 (`riscv32imc-unknown-none-elf`), but ultimately, the error is the same (unsupported target).

Am I missing something here, or is there something else that needs to be done to get no_std working?

Vollbrecht, 2023-12-22:
you are mixing two completely unrelated things her. Yes esp-idf-sys has a feature called no_std but this has nothing to do with the esp-hal no_std project. If you are using esp-idf-sys you always using esp-idf. You can run a no_std rust application on it but that doesn't change the fact that the underlying rust target is still the normal one against esp-idf. You cannot use esp-idf-sys against the rust no_std targets

thedevleon, 2023-12-22:
I see, thanks for the clarification!

closed 2023-12-22 by thedevleon