taks/esp32-nimble #63: ESP32-C3 crashes when 2nd device connects in `ble_server` example

ChocolateLoverRaj, 2024-01-03:
Steps to reproduce:
- Change `.cargo/config.toml` to be for ESP32-C3
- Change the command from `espflash --monitor` to `espflash flash --monitor`
- `cargo run --example ble_server`
- Connect to the ESP32-C3 with a device (I used [LightBlue](https://play.google.com/store/apps/details?id=com.punchthrough.lightblueexplorer)
- Connect to the ESP32-C3 with another device while the first one is still connected

Error:
```
thread '<unnamed>' panicked at examples/ble_server.rs:20:42:
called `Result::unwrap()` on an `Err` value: Operation failed due to resource exhaustion.
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x4202b1f1 on core 0
```

Full log (the first part was not included because I reached the terminal max history):
```
I (43) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (43) boot: compile time Jun  7 2023 07:59:10
I (44) boot: chip revision: v0.4
I (48) boot.esp32c3: SPI Speed      : 40MHz
I (53) boot.esp32c3: SPI Mode       : DIO
I (57) boot.esp32c3: SPI Flash Size : 4MB
I (62) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (71) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (86) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 003f0000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=2f650h (194128) map
I (156) esp_image: segment 1: paddr=0003f678 vaddr=3fc91200 size=009a0h (  2464) load
I (157) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=8836ch (557932) map
I (284) esp_image: segment 3: paddr=000c8394 vaddr=3fc91ba0 size=01894h (  6292) load
I (286) esp_image: segment 4: paddr=000c9c30 vaddr=40380000 size=11198h ( 70040) load
I (312) boot: Loaded app from partition at offset 0x10000
I (312) boot: Disabling RNG early entropy source...
I (324) cpu_start: Unicore app
I (324) cpu_start: Pro cpu up.
I (333) cpu_start: Pro cpu start user code
I (333) cpu_start: cpu freq: 160000000 Hz
I (333) cpu_start: Application information:
I (336) cpu_start: Project name:     libespidf
I (341) cpu_start: App version:      v0.4.0-25-g75bb3e8-dirty
I (348) cpu_start: Compile time:     Jan  3 2024 13:10:34
I (354) cpu_start: ELF file SHA256:  0000000000000000...
I (360) cpu_start: ESP-IDF:          v5.1.1
I (364) cpu_start: Min chip rev:     v0.3
I (369) cpu_start: Max chip rev:     v0.99 
I (374) cpu_start: Chip rev:         v0.4
I (379) heap_init: Initializing. RAM available for dynamic allocation:
I (386) heap_init: At 3FC95ED0 len 00046840 (282 KiB): DRAM
I (392) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (399) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (406) spi_flash: detected chip: generic
I (410) spi_flash: flash io: dio
W (414) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (424) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (433) sleep: Configure to isolate all GPIO pins in sleep state
I (440) sleep: Enable automatic switching of GPIO sleep configuration
I (447) coexist: coex firmware version: 80b0d89
I (452) coexist: coexist rom version 9387209
I (458) app_start: Starting scheduler on CPU0
I (462) main_task: Started on CPU0
I (462) main_task: Calling app_main()
I (472) BLE_INIT: BT controller compile version [963cad4]
I (472) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (522) BLE_INIT: Bluetooth MAC: ec:da:3b:35:6e:1a

[... 926 lines cut ...]
         handle 15
   min_key_size 0
          flags [READ|WRITE]
characteristic
           uuid 3c9a3f00-8ed3-4bdf-8a39-a01bebede295
     def_handle 16
     val_handle 17
   min_key_size 0
          flags [READ|WRITE]
I (652) NimBLE: GAP procedure initiated: advertise; 
I (652) NimBLE: disc_mode=2
I (662) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (672) NimBLE: 

I (732) ble_server: Client connected
I (732) NimBLE: GAP procedure initiated: 
I (732) NimBLE: connection parameter update; conn_handle=1 itvl_min=24 itvl_max=48 latency=0 supervision_timeout=60 min_ce_len=0 max_ce_len=0
I (742) NimBLE: 

I (752) ble_server: Multi-connect support: start advertising
I (762) NimBLE: GAP procedure initiated: advertise; 
I (762) NimBLE: disc_mode=2
I (762) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (772) NimBLE: 
```

oleander, 2024-01-10:
> Operation failed due to resource exhaustion.

You likely need to increase maximum number of concurrent connections. The connection count before the crash is 3, try increasing it to 9 using "menuconfig" or by updating the sdkconfig directly. 

Also try increasing the stack size to 7000.

ChocolateLoverRaj, 2024-01-10:
@oleander thanks for the tip.

I found [this option](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-bt-nimble-max-connections), and I changed the `sdkconfig.defaults` file and added this:
```
CONFIG_BT_CONTROLLER_ENABLED=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=9
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=9
```
and now I'm able to connect 4 devices without the microcontroller crashing. Should I make a PR adding these options?

PR #68 (2024-01-11): Fixed crash when max connections reached
Fix #63
closed 2024-01-11 by taks