taks/esp32-nimble #65: How to turn off logging / output to `stdout`?

ChocolateLoverRaj, 2024-01-09:
I am going to use `stdout` for sending data, and I want to use this library without interfering with other messages sent to `stdout`. Is there an option to turn off all logging that this library does?

taks, 2024-01-10:
Log output can be turned off by adding the following code.

```rust
esp_idf_svc::log::EspLogger.set_target_level("*", log::LevelFilter::Off);
```

ChocolateLoverRaj, 2024-01-10:
@taks I tried https://github.com/taks/esp32-nimble/issues/65#issuecomment-1884141975 and it got rid of the green / info messages. I still get this output though:
```
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 9
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid a78b10d1-0ef3-454f-8ddd-b09fb08fdbe6
         handle 10
     end_handle 13
characteristic
           uuid f935f617-1ebf-4b72-9ce8-732c2d7f531c
     def_handle 11
     val_handle 12
   min_key_size 0
          flags [READ|WRITE|NOTIFY]
ccc descriptor
           uuid 0x2902
         handle 13
   min_key_size 0
          flags [READ|WRITE]
```

taks, 2024-01-10:
Please use the latest version.
If you are using the latest version, please delete `server.ble_gatts_show_local();` code.

ChocolateLoverRaj, 2024-01-10:
Removing `server.ble_gatts_show_local()` got rid of that output. My question is answered, thank you @taks .

closed 2024-01-10 by ChocolateLoverRaj