taks/esp32-nimble #72: Mark `&mut BLEAdvertising` as `Send`?

ChocolateLoverRaj, 2024-01-22:
Is it safe for `&mut BLEAdvertising` to be marked with `Send` similar to https://github.com/taks/esp32-nimble/commit/046882663252b08626744cb1ee3d9fffa5ef1fa2? I want to access the advertising from the `on_write` callback. Currently I'm using channels, but it would be more convenient if I could just use `Arc`.

This is related to #17

taks, 2024-01-25:
Is it possible to paste the main.rs source code?

ChocolateLoverRaj, 2024-01-25:
> Is it possible to paste the main.rs source code?

Here is the code which produces that error:
https://github.com/ChocolateLoverRaj/rust-esp32c3-examples/blob/ae197aef46464ed5950683d3d34ae638a13e2c79/sensor-connect/src/main.rs#L132

it's not a _minimum_ example. Let me know if I should make a minimum example.

ChocolateLoverRaj, 2024-01-25:
I found out a much simpler way to reproduce the error: https://github.com/ChocolateLoverRaj/esp32-nimble/blob/d56af552fe6f8def6c951012b18c08ebff5dea03/examples/ble_secure_server.rs#L25

taks, 2024-01-26:
> Here is the code which produces that error: https://github.com/ChocolateLoverRaj/rust-esp32c3-examples/blob/ae197aef46464ed5950683d3d34ae638a13e2c79/sensor-connect/src/main.rs#L132
>
> it's not a [...]

I corrected the error in the project you attached.
Please check it.
https://github.com/taks/rust-esp32c3-examples/tree/fix-build-error/sensor-connect

> I found out a much simpler way to reproduce the error: https://github.com/ChocolateLoverRaj/esp32-nimble/blob/d56af552fe6f8def6c951012b18c08ebff5dea03/examples/ble_secure_server.rs#L25

It should work if you change `_server` to `server` on line 16.

ChocolateLoverRaj, 2024-01-26:
Thanks for fixing the error. I noticed that you used `esp32_nimble::utilities::mutex::Mutex` instead of `std::sync::Mutex`. Why?

PR #73 (2024-01-24): Mark BLEAdvertising as Send
Fix #72
closed 2024-07-27 by ChocolateLoverRaj