taks/esp32-nimble #78: `BLEDevice::set_device_name` doesn't change the name

ChocolateLoverRaj, 2024-01-29:
This is what I tried: https://github.com/ChocolateLoverRaj/esp32-nimble/commit/746d2bcfcbacae0001b8acfbf44b2e81a053374d
```rs
let name = String::from_utf8(args.recv_data.to_vec())
    .unwrap()
    .trim()
    .to_owned();
println!("Changing name to: {:#?}", name);
BLEDevice::set_device_name(&name).unwrap();
```
The name doesn't change. I also tried starting advertising again after that, but the name still doesn't change.

taks, 2024-01-31:
Please try the following code.
```rust
      if let Ok(name) = std::str::from_utf8(args.recv_data()) {
        let mut ble_advertising = ble_advertising.lock();
        ble_advertising.name(name);
        ble_advertising.stop().unwrap();
        ble_advertising.start().unwrap();
      }
```

ChocolateLoverRaj, 2024-02-01:
I'm not sure what type `ble_advertising` is in the code `ble_advertising.lock()`. 

This is what I tried instead:
```
let advertising = BLEDevice::take().get_advertising();
advertising.name(&name);
advertising.stop().unwrap();
advertising.start().unwrap();
```
And then this what happened to the name:
```
Name: �^�?�^�?TT-Serve�
Address: EC:DA:3B:35:6E:1A
RSSI: -54 dBm
Last advertisement packet:
Raw data: 0x0201061107FAFAFAFAFAFAFAFAFAFAFAFAFAFAFAFA08094E65774E616D651209E45EC93FE45EC93F54542D536572766580
```

ChocolateLoverRaj, 2024-02-01:
Actually I need to pull the latest changes first and then try again

taks, 2024-02-01:
If the length of the overwrite string is short, the name is moved from the scan response data to the advertising data.
In this case, the scan response data was not cleared and the display becomes incorrect.

I have fixed this bug in the [fix-advertising](https://github.com/taks/esp32-nimble/tree/fix-advertising) branch.
Please check.

ChocolateLoverRaj, 2024-02-01:
I tried the same code as https://github.com/taks/esp32-nimble/issues/78#issuecomment-1920356317 in the `fix-advertising` branch and it works.

PR #85 (2024-02-01): Fix advertising
Fix #78
closed 2024-02-01 by taks