taks/esp32-nimble #88: Passkey Size

sampaioletti, 2024-02-16:
Would it be worth adding documentation that the minimum passkey length is 6 digits for BLE

See here for reference
[https://esp32.com/viewtopic.php?t=13124](https://esp32.com/viewtopic.php?t=13124)

So if you set a passkey to say '1234' the passkey will actually be '001234' I'm not sure if that's guaranteed or where that transform occurs, but that is how it currently works. I was struggling with it for a bit, may not be alone. (:

taks, 2024-02-16:
Yes, please.

ChocolateLoverRaj, 2024-02-26:
> I'm not sure if that's guaranteed or where that transform occurs

FYI, I'm pretty sure the transform occurs when the number gets converted to `u32` and then the `u32` gets converted into bytes.

I also got confused when I found out that 1234 turns into "001234". You're not the only one.

PR #89 (2024-02-28): Update README.md with Passkey details
fixes #88
commit ed78633b (2024-02-28): Update README.md with Passkey details

fixes #88
closed 2024-02-28 by taks