taks/esp32-nimble #99: Cleanup BLEServer on BLEDevice::deinit

mbv, 2024-02-21:
I’m in the process of developing an application that has the capability to establish independent BLE servers in various sections. Only one server is active at any given time. When a section ceases its server, it invokes `BLEDevice::deinit`. The expectation is that the BLEServer is reset, but in reality, `BLEDevice::deinit` only resets the started flag on the BLEServer, leaving all services and callbacks intact.

I propose the introduction of a `cleanup_server` flag to `BLEDevice::deinit` that would ensure a thorough cleanup of all elements within the BLEServer.

taks, 2024-02-22:
I added the `BLEDevice::deinit_full` function in the [deinit_full](https://github.com/taks/esp32-nimble/tree/deinit_full) branch. 
Please check.

mbv, 2024-02-22:
@taks thank you, it passed my tests.

PR #100 (2024-02-22): Added BLEDevice::deinit_full
Fix #99
closed 2024-02-22 by taks