taks/esp32-nimble #115: Support L2CAP

milewski, 2024-03-25:
Hi any plan to support l2cap?

taks, 2024-03-28:
There are no immediate plans to support l2cap.
PR is welcome.

taks, 2024-04-02:
I have implemented a simple feature of L2CAP.
Are there any additional functions required?

taks, 2024-04-02:
The ps3 controller does not seem to use BLE communication, but bluetooth classic communication.
Unfortunately, the nimble library is available for BLE only, so it cannot be used for communication with ps3 controllers.

milewski, 2024-04-03:
Yeah, that was what I found as well. None of the newer ESP chips worked either because they don't support classic Bluetooth. Only an old ESP32 worked out. 

I have ordered a ps5 controller which does support BLE, I found this: https://github.com/rodneybakiskan/ps5-esp32 once I get the controller I will try to port the classic code to the ble version and see how it goes.. l2cap still required

taks, 2024-04-03:
The ps5 controller also seem to use bluetooth classic communication.
https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#bluetooth-gamepads-and-their-protocol

PR #117 (2024-09-30): Added L2CAP support
Fix #115
closed 2024-09-30 by taks