taks/esp32-nimble #119: ble_keyboard example doesn't work

HaoboGu, 2024-04-03:
Hello! I'm trying ble_keyboard example, the advertising and connection is good. But I cannot receive any content after connected to my mac. Here is the log:

```
warning: `esp32-nimble` (lib) generated 1 warning
    Finished dev [optimized + debuginfo] target(s) in 5.44s
     Running `espflash flash --monitor target/riscv32imc-esp-espidf/debug/examples/ble_keyboard`
[2024-04-03T10:42:44Z INFO ] Detected 2 serial ports
[2024-04-03T10:42:44Z INFO ] Ports which match a known common dev board are highlighted
[2024-04-03T10:42:44Z INFO ] Please select a port
[2024-04-03T10:42:55Z INFO ] Serial port: '/dev/cu.usbmodem211301'
[2024-04-03T10:42:55Z INFO ] Connecting...
[2024-04-03T10:42:55Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.4)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       ec:da:3b:d1:87:c4
App/part. size:    839,200/4,128,768 bytes, 20.33%
[2024-04-03T10:42:55Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-04-03T10:42:55Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:04] [========================================]     455/455     0x10000                                                                                                                                [2024-04-03T10:43:01Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380632
0x40380632 - spiflash_start_default
    at /Users/haobogu/Projects/rust/esp32-nimble/.embuild/espressif/esp-idf/v5.1.2/components/spi_flash/esp_flash_api.c:194
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1714
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (19) boot: ESP-IDF v5.1.2-342-gbcf1645e44 2nd stage bootloader
I (20) boot: compile time Dec 12 2023 10:50:58
I (20) boot: chip revision: v0.4
I (24) boot.esp32c3: SPI Speed      : 40MHz
I (29) boot.esp32c3: SPI Mode       : DIO
I (33) boot.esp32c3: SPI Flash Size : 4MB
I (38) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (69) boot:  2 factory          factory app      00 00 00010000 003f0000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=30098h (196760) map
I (133) esp_image: segment 1: paddr=000400c0 vaddr=3fc90400 size=0231ch (  8988) load
I (136) esp_image: segment 2: paddr=000423e4 vaddr=40380000 size=0dc34h ( 56372) load
I (153) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=8a634h (566836) map
I (279) esp_image: segment 4: paddr=000da65c vaddr=4038dc34 size=0279ch ( 10140) load
I (287) boot: Loaded app from partition at offset 0x10000
I (287) boot: Disabling RNG early entropy source...
I (299) cpu_start: Unicore app
I (299) cpu_start: Pro cpu up.
I (309) cpu_start: Pro cpu start user code
I (309) cpu_start: cpu freq: 160000000 Hz
I (309) cpu_start: Application information:
I (312) cpu_start: Project name:     libespidf
I (317) cpu_start: App version:      v0.6.0-14-g1e9c439-dirty
I (324) cpu_start: Compile time:     Apr  3 2024 18:41:34
I (330) cpu_start: ELF file SHA256:  0000000000000000...
I (336) cpu_start: ESP-IDF:          v5.1.2
I (341) cpu_start: Min chip rev:     v0.3
I (345) cpu_start: Max chip rev:     v0.99 
I (350) cpu_start: Chip rev:         v0.4
I (355) heap_init: Initializing. RAM available for dynamic allocation:
I (362) heap_init: At 3FC950B0 len 0002AF50 (171 KiB): DRAM
I (368) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (375) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (383) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (391) spi_flash: detected chip: generic
I (394) spi_flash: flash io: dio
W (398) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (407) sleep: Configure to isolate all GPIO pins in sleep state
I (413) sleep: Enable automatic switching of GPIO sleep configuration
I (420) coexist: coex firmware version: b6d5e8c
I (426) coexist: coexist rom version 9387209
I (431) app_start: Starting scheduler on CPU0
I (435) main_task: Started on CPU0
I (435) main_task: Calling app_main()
I (445) BLE_INIT: BT controller compile version [59725b5]
I (445) BLE_INIT: Bluetooth MAC: ec:da:3b:d1:87:c6

I (455) phy_init: phy_version 1130,b4e4b80,Sep  5 2023,11:09:30
I (495) esp32_nimble::ble_device: BLE Host Task Started
I (505) NimBLE: GAP procedure initiated: stop advertising.

I (505) esp32_nimble::ble_device: Device Address: EC:DA:3B:D1:87:C6
I (515) NimBLE: GAP procedure initiated: advertise; 
I (515) NimBLE: disc_mode=2
I (515) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (525) NimBLE: 

I (32775) esp32_nimble::server::ble_server: mtu update event; conn_handle=1 mtu=256
I (35525) ble_keyboard: Sending 'Hello world'...
I (40685) ble_keyboard: Sending 'Hello world'...
I (45845) ble_keyboard: Sending 'Hello world'...
I (51005) ble_keyboard: Sending 'Hello world'...
I (56165) ble_keyboard: Sending 'Hello world'...
```

Appreciate if anyone can help

taks, 2024-04-04:
Could you try adding `resolve_rpa()` to security?
```rust
    device
      .security()
      .set_auth(AuthReq::all())
      .set_io_cap(SecurityIOCap::NoInputNoOutput);
      .resolve_rpa();
```

HaoboGu, 2024-04-04:
Yeah, that works! I opened a PR adding this to keyboard example

closed 2024-04-04 by taks