taks/esp32-nimble #125: ble_gatt_chr_def, Required fields/characteristics in struct?

terelldunn, 2024-05-21:
I am new to Rust and not very knowledgeable about programming in general so bear with me.

I am using an example from the [ApolloLabs ESP32C3 repository](https://github.com/apollolabsdev/ESP32C3/tree/main/std_bt_examples/ble-server).

The compiler seems to want a "cpfd" field or characteristic in the ble_gatt_chr_def struct. 

[image]

I don't see that field listed [here](https://docs.rs/apache-nimble-sys/latest/apache_nimble_sys/struct.ble_gatt_chr_def.html).

I have tried adding it to the struct and assigning various values or even variations of Option<>, but the compiler gives a few different errors after that point. Again, I have no idea what I'm doing to be honest.

Google is no help and doesn't return much in regards to "cpfd".

I'm compiling with 5.2.1.

My target is correct. "riscv32imc-esp-espidf"

The following is present in sdkconfig.
CONFIG_BT_ENABLED=y
CONFIG_BT_BLE_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y

Any ideas?

terelldunn, 2024-05-21:
It looks like it may be in Volume 3, Part G, Chapter 3.3.3.5 and 3.3.3.6 of the Bluetooth Core specification v5.3. "Characteristic Presentation Format Declaration". I completely stole the info pointing to the exact bluetooth spec from another forum. I won't take credit there.

taks, 2024-05-21:
This issue is the same as in #110, #123.
It was fixed in the `main` branch but not released as a new version.

I have just released version 0.6.1 which fixes this issue.
Please check.

closed 2024-05-23 by taks