taks/esp32-nimble #132: How do handle large reads (larger than MAX MTU)?

juliankrieger, 2024-07-12:
How would I go on about handling reading values that are larger than the maximum MTU from a characteristic? My characteristic just returns a value of "0" if I write a large string into it.

taks, 2024-07-14:
Please provide the following information:
- Peripheral (server) device
- Central (client) device
- The length of string
- Can you read the value of characteristic using android or iOS?

juliankrieger, 2024-07-14:
- Peripheral: ESP32 S3
- Central: Samsung Galaxy S8
- 3900 bytes
- No. 

My current WIP solution writes MTU - 3 bytes onto read‘s value inside the on_read callback, therefore sending the data in chunks.

taks, 2024-07-15:
> My current WIP solution writes MTU - 3 bytes onto read‘s value inside the on_read callback, therefore sending the data in chunks.
Could you paste the code?

taks, 2024-07-17:
I checked the specifications, the maximum length of characteristic appears to be 512 bytes.
([3.2.9 of the specification](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Core-54/out/en/host/attribute-protocol--att-.html#UUID-d24e0156-618e-728d-f19b-01349ced952b))

I have checked communication from esp32 to esp32.
It was possible to read 800 bytes larger than the specification.
But when I increased the size to 1600 bytes, the connection was broken during the read.

taks, 2024-09-24:
Thanks.
You are dividing the data with a length of `mtu() - 3`.
But you don't need to consider the mtu size in `on_write` and `on_read` because the combination and division are handled on the nimble side.

※ If the data size exceeds `mtu() - 3`, `on_write` is called only once, but `on_read` is called multiple times.

juliankrieger, 2024-09-25:
Can you explain a bit further how this works? You say combination and division are handles on the nimble side, but how do I get the entire data in on_read and on_write?

juliankrieger, 2024-09-25:
A follow up question: Would it be possible to support GATT long writes (Prepare Write and Write Execute Events) and long reads (read Blob event). Do I even need this if I want to write and read large data? I would be willing to open a PR for this and I'm hoping you could give me a few pointers on where to start.

juliankrieger, 2024-09-25:
Through some research, I have learned that the bluetooth specification sets a maximum characteristic length of 512 bytes. So this is what my original Goal is and was with the code above. Reading and writing characteristics that are longer than that.

taks, 2024-09-25:
I found an answer at stack overflow.

https://stackoverflow.com/questions/41351913/best-approach-for-transfering-large-data-chunks-over-ble