taks/esp32-nimble #136: Memory Leak When Looping Scan

Ticlext-Altihaf, 2024-07-23:
## How
Chip type:         esp32 (revision v3.1)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
App/part. size:    642,848/4,128,768 bytes, 15.57%
ESP-IDF:          v5.2.2
Min chip rev:     v0.0
Max chip rev:     v3.99 
Chip rev:         v3.1

```rs
fn main() -> Result<()> {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();
    // `async-io` uses the ESP IDF `eventfd` syscall to implement async IO.
    // If you use `tokio`, you still have to do the same as it also uses the `eventfd` syscall
    esp_idf_svc::io::vfs::initialize_eventfd(5).unwrap();

    tokio::runtime::Builder::new_current_thread()
        .enable_all()
        .build()
        .unwrap()
        .block_on(async move {
            loop {
                let ble_device = BLEDevice::take();
                let ble_scan = ble_device.get_scan();
                ble_scan
                    .active_scan(true)
                    .interval(100)
                    .window(99)
                    .on_result(move |_scan, device| {
                       
                    });

                ble_scan.start(10000).await.unwrap();
                log::warn!("esp_get_minimum_free_heap_size: {:?}", unsafe {
                    esp_idf_sys::esp_get_minimum_free_heap_size()
                });
                log::warn!("esp_get_free_heap_size: {:?}", unsafe {
                    esp_idf_sys::esp_get_free_heap_size()
                });
            }
        });
    return Ok(());
 ```

## Result

```
I (1069) esp32_nimble::client::ble_advertised_device: Unhandled type: adType: 0x1B
W (2939) esp_idf_example: esp_get_minimum_free_heap_size: 132340
W (2939) esp_idf_example: esp_get_free_heap_size: 132408
I (2939) NimBLE: GAP procedure initiated: discovery; 
I (2939) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1 
I (2949) NimBLE: duration=2000ms
I (2959) NimBLE: 

... sometime later

I (1124849) esp32_nimble::client::ble_advertised_device: Unhandled type: adType: 0x1B
W (1134689) esp_idf_example: esp_get_minimum_free_heap_size: 72468
W (1134689) esp_idf_example: esp_get_free_heap_size: 74084
I (1134689) NimBLE: GAP procedure initiated: discovery; 
I (1134699) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1 
I (1134709) NimBLE: duration=10000ms
I (1134709) NimBLE: 

```

## Expected

It should not keep hogging ram, or did I do something wrong?

taks, 2024-07-23:
Could you try calling `ble_scan.clear_results()` in each loop?

Ticlext-Altihaf, 2024-07-23:
It seems to solve it, sorry for bothering you, thanks

closed 2024-07-23 by Ticlext-Altihaf