taks/esp32-nimble #143: failed to receive data from characteristic indication

ricson-hoo, 2024-09-26:
Hello, I am working on a blood pressure monitor. I followed the example at https://github.com/taks/esp32-nimble/blob/main/examples/ble_client.rs to write my code for it, but it doesn't work. The subscribe_indicate function is called, but the on_notify callback is never triggered.

My code looks like this:

....

                let mut client = BLEClient::new();
                client.on_connect(|client| {
                  client.update_conn_params(120, 120, 0, 60).unwrap();
                });
                client.connect(&device.addr()).await;
          
                let service = client
                  .get_service(uuid128!(self::YUELL_BLOOD_PRESSURE_MONITOR_SERVICE_UUID))
                  .await.unwrap();
          
                let characteristic_uuid = uuid128!(self::YUELL_BLOOD_PRESSURE_MONITOR_CHARACTERISTIC_UUID);
                let characteristic = service.get_characteristic(characteristic_uuid).await.unwrap();
                if !characteristic.can_indicate() {
                    log::error!("characteristic can't indicate: {}", characteristic);
                    return;
                  }

                characteristic
                .on_notify(|data| {
                    log::info!("recevied data from characteristic");
                    log::info!("received data from bpm：{}", core::str::from_utf8(data).unwrap());
                });

                characteristic.subscribe_indicate(true).await.expect("failed subscribe_indicate");

                log::info!("after subscribe_indicate {}", characteristic);

                time::sleep(time::Duration::from_secs(10000)).await;

.....

Here are the logs output:

I (4326) smart_health_kit::peripheral_client::yuwell_bpm: Yuwell BP-YE670A found, mac addr 44:28:A3:60:87:F8!
I (4326) smart_health_kit::peripheral_client::yuwell_bpm: after device found!
I (4336) NimBLE: GAP procedure initiated: connect; 
I (4336) NimBLE: peer_addr_type=0 peer_addr=
I (4346) NimBLE: 44:28:a3:60:87:f8
I (4346) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (4356) NimBLE: 

I (4936) NimBLE: GATT procedure initiated: exchange mtu

I (5006) esp32_nimble::client::ble_client: mtu update event; conn_handle=0 mtu=247
I (5006) NimBLE: GAP procedure initiated: 
I (5006) NimBLE: connection parameter update; conn_handle=0 itvl_min=120 itvl_max=120 latency=0 supervision_timeout=60 min_ce_len=0 max_ce_len=0
I (5016) NimBLE: 

I (5026) NimBLE: GATT procedure initiated: discover all services

I (5256) NimBLE: GATT procedure initiated: discover all characteristics; 
I (5256) NimBLE: start_handle=49 end_handle=67

I (5806) NimBLE: GATT procedure initiated: discover all characteristics; 
I (5806) NimBLE: start_handle=51 end_handle=67

I (6106) NimBLE: GATT procedure initiated: discover all descriptors; 
I (6106) NimBLE: chr_val_handle=51 end_handle=53

I (6406) NimBLE: GATT procedure initiated: write; 
I (6406) NimBLE: att_handle=52 len=2

I (6706) smart_health_kit::peripheral_client::yuwell_bpm: after subscribe_indicate BLERemoteCharacteristic[0x2a35]

I tried to output a debug log on ble_client.rs as followed, but no log was output at all:

 extern "C" fn handle_gap_event(event: *mut esp_idf_sys::ble_gap_event, arg: *mut c_void) -> i32 {
         ....
         BLE_GAP_EVENT_NOTIFY_RX => {

        let notify_rx = unsafe { &event.__bindgen_anon_1.notify_rx };

        ::log::info!(
          "notify_rx={:#?}",
          notify_rx
        ); 

===================

by the way, the blood pressure monitor works on micropython program using below code:

    async def connection_and_monitor_task(self):
        while True:
            if self.device and (not self.connection or not self.connection.is_connected()):
                try:
                    print("YuWellBpMonitorService:Connecting to", self.device)
                    self.connection = await self.device.connect()
                    beepUtil.ok_notice()
                    async with self.connection:
                        try:
                            bp_service = await self.connection.service(self._ENV_BLOOD_PRESSURE_UUID)
                            bp_characteristic = await bp_service.characteristic(self._ENV_BLOOD_PRESSURE_VALUE_UUID)
                            
                            await bp_characteristic.subscribe(notify=False, indicate=True)

                            print("YuWellBpMonitorService:connection done")

                            while self.connection and self.connection.is_connected():
                                try:
                                    print("YuWellBpMonitorService:waiting for indication")
                                    data = await bp_characteristic.indicated()
                                    parsed_data = self._decode_data(data)
                                    timestamp = trueTime()
                                    print("timestamp:%s"%timestamp)
                                    uuid = "%s|%d" % (self.deviceId, timestamp)
                                    .....

I am a newbie on ESP32 with Rust. Any help is appreciated, thanks.

ricson-hoo, 2024-09-26:
I tried to downgrade to version 0.7 with following code, and it works.

let mut client = BLEClient::new();
            client.on_connect(|client| {
              //client.update_conn_params(120, 120, 0, 60).unwrap();
              log::info!("connected to ble server...")
            });
            client.connect(&device.addr()).await;
      
            let service = client
              .get_service(uuid128!(self::YUELL_BLOOD_PRESSURE_MONITOR_SERVICE_UUID))
              .await.unwrap();
      
            let characteristic_uuid = uuid128!(self::YUELL_BLOOD_PRESSURE_MONITOR_CHARACTERISTIC_UUID);
            let characteristic = service.get_characteristic(characteristic_uuid).await.unwrap();
            if !characteristic.can_indicate() {
                log::error!("characteristic can't indicate: {}", characteristic);
                return;
              }

            characteristic
            .on_notify(|data| {
                log::info!("recevied data from characteristic,{:?}",data);
                //log::info!("received data from bpm：{}", core::str::from_utf8(data).unwrap());
            });

            characteristic.subscribe_indicate(true).await.expect("failed subscribe_indicate");

            log::info!("subscribe_indicate called {}", characteristic);

=================================
 logs output:

I (3839) smart_health_kit::peripheral_client::yuwell_bpm: connected to ble server...
I (3849) NimBLE: GATT procedure initiated: discover all services

I (4039) NimBLE: GATT procedure initiated: discover all characteristics; 
I (4039) NimBLE: start_handle=49 end_handle=67

I (4239) NimBLE: GATT procedure initiated: discover all characteristics; 
I (4239) NimBLE: start_handle=51 end_handle=67

I (4339) NimBLE: GATT procedure initiated: discover all descriptors; 
I (4339) NimBLE: chr_val_handle=51 end_handle=53

I (4439) NimBLE: GATT procedure initiated: write; 
I (4439) NimBLE: att_handle=52 len=2

I (4539) smart_health_kit::peripheral_client::yuwell_bpm: subscribe_indicate called BLERemoteCharacteristic[0x2a35]
I (47589) smart_health_kit::peripheral_client::yuwell_bpm: recevied data from characteristic,[30, 119, 0, 84, 0, 95, 0, 229, 7, 12, 14, 7, 39, 27, 74, 0, 0, 2, 0]

taks, 2024-09-27:
Thanks @ricson-hoo.
I fixed this bug in the [fix-multiple-drop](https://github.com/taks/esp32-nimble/tree/fix-multiple-drop) branch.
Could you check this?

taks, 2024-09-27:
Instead of rewriting the files in the .cargo directory, please rewrite `esp32-nimble = "0.8.0"` to 
```
esp32-nimble = { git = "https://github.com/taks/esp32-nimble.git", branch = "fix-multiple-drop" }
```
in your Cargo.toml.

ricson-hoo, 2024-09-27:
Hi, taks, thanks.

This time, I removed /Users/ricson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.8.0 and changed the Cargo.toml configuration from esp32-nimble = "0.8.0" to esp32-nimble = { git = "https://github.com/taks/esp32-nimble.git", branch = "fix-multiple-drop" }. However, after running it, the result remains the same.

I noticed that the code has now been pulled to: /Users/ricson/.cargo/git/checkouts/esp32-nimble-fc9d2c54a3bfa7fe/f3e6d96

ricson-hoo, 2024-09-27:
> It appears that the `time::sleep` code that was in the first attached code is missing
> 
> ```rust
> time::sleep(time::Duration::from_secs(10000)).await;
> ```

Oh yes, I moved it out of that code block. It seems this caused the characteristic to end its life immediately after subscribe_indicate. 

I moved it back, and finally it works.

Now it logs:
I (3795) xite_iot_gateway::peripheral_client::yuwell_bpm: characteristic.subscribe_indicate called: BLERemoteCharacteristic[0x2a35]
I (40745) xite_iot_gateway::peripheral_client::yuwell_bpm: received data from characteristic, [30, 106, 0, 73, 0, 84, 0, 229, 7, 12, 15, 20, 24, 22, 65, 0, 0, 0, 0]
I (158945) esp32_nimble::client::ble_client: Disconnected: Err(Remote User Terminated Connection)

Thank you so much for your help!

PR #144 (2024-09-28): Fixed BLEClient dropping multiple times
Fix #143
commit f3e6d963 (2024-09-28): Fixed BLEClient dropping multiple times (Fix #143)
closed 2024-09-28 by taks