taks/esp32-nimble #147: Build issue for v0.8.1

jacksonemiller, 2024-10-21:
Getting these errors when trying to build. I had previously been on v0.8.0 with no build issues. 
[build-output.txt](https://github.com/user-attachments/files/17452919/build-output.txt)
[Cargo.txt](https://github.com/user-attachments/files/17452923/Cargo.txt)
[sdkconfig.txt](https://github.com/user-attachments/files/17452926/sdkconfig.txt)

taks, 2024-10-22:
Could you also attach your `.cargo/config.toml`?

jacksonemiller, 2024-10-22:
Sure, thanks for the reply.
[config.txt](https://github.com/user-attachments/files/17478904/config.txt)

taks, 2024-10-23:
Thanks @jacksonemiller .
I fixed this issue in the [fix-esp-idf-v5.2.2](https://github.com/taks/esp32-nimble/tree/fix-esp-idf-v5.2.2) branch.
Could you check it?

vermasrijan19, 2024-10-23:
I am trying to build with IDF version v5.2.2 and am facing this error : 

`error[E0432]: unresolved imports `esp_idf_sys::ble_uuid_any_t`, `esp_idf_sys::ble_uuid_cmp`
 --> /home/srijan/.cargo/git/checkouts/esp32-nimble-fc9d2c54a3bfa7fe/980873d/src/server/ble_characteristic.rs:5:19
  |
5 | use esp_idf_sys::{ble_uuid_any_t, ble_uuid_cmp};
  |                   ^^^^^^^^^^^^^^  ^^^^^^^^^^^^ no `ble_uuid_cmp` in `sys`
  |                   |
  |                   no `ble_uuid_any_t` in `sys`
`

am I missing something?

jacksonemiller, 2024-10-25:
Working like a charm so far, I appreciate it greatly.

PR #148 (2024-10-25): Fix ESP-IDF v5.2.2 build
Fix #147
closed 2024-10-25 by jacksonemiller