taks/esp32-nimble #151: Getting stuck in notify after a while

vermasrijan19, 2024-11-11:
I am using the ble to transmit constant sensor data, after running for a while, there are some Error and then code seemingly gets stuck in in notify loop.

All callbacks are working as expected but the loop in which i am transmitting data is stuck.

```
I (183351) walk_rs::initialize_components: SuccessNotify
I (183360) NimBLE: GATT procedure initiated: notify;
I (183363) NimBLE: att_handle=18

I (183367) walk_rs::initialize_components: ErrorGatt
I (183375) NimBLE: GATT procedure initiated: notify;
I (183378) NimBLE: att_handle=18

I (183382) walk_rs::initialize_components: ErrorGatt
I (183391) NimBLE: GATT procedure initiated: notify;
I (183393) NimBLE: att_handle=18

I (183398) walk_rs::initialize_components: ErrorGatt
I (183406) NimBLE: GATT procedure initiated: notify;
I (183409) NimBLE: att_handle=18

I (183413) walk_rs::initialize_components: ErrorGatt
I (183426) NimBLE: GATT procedure initiated: notify;
I (183426) NimBLE: att_handle=18

I (183429) walk_rs::initialize_components: SuccessNotify
I (183438) NimBLE: GATT procedure initiated: notify;
I (183440) NimBLE: att_handle=18

I (183445) walk_rs::initialize_components: SuccessNotify
I (183453) NimBLE: GATT procedure initiated: notify;
I (183456) NimBLE: att_handle=18

I (183460) walk_rs::initialize_components: SuccessNotify
I (183469) NimBLE: GATT procedure initiated: notify;
I (183472) NimBLE: att_handle=18

I (183476) walk_rs::initialize_components: SuccessNotify
I (183485) NimBLE: GATT procedure initiated: notify;
I (183488) NimBLE: att_handle=18

I (183492) walk_rs::initialize_components: SuccessNotify
I (183501) NimBLE: GATT procedure initiated: notify;
I (183503) NimBLE: att_handle=18

I (183508) walk_rs::initialize_components: SuccessNotify
I (183516) NimBLE: GATT procedure initiated: notify;
I (183519) NimBLE: att_handle=18

I (183524) walk_rs::initialize_components: SuccessNotify
I (183533) NimBLE: GATT procedure initiated: notify;
I (183535) NimBLE: att_handle=18

I (222177) walk_rs::initialize_components: Client disconnected (Err(Remote User Terminated Connection))
```

vermasrijan19, 2024-11-16:
After debugging for a while, it seems that it gets stuck in `handle_gap_event` in  ble_characteristic.rs when the following code is executed, there seems to be a n event with ctxt.op as 0.

`    let mutex = unsafe { voidp_to_ref::<Mutex<Self>>(arg) };
    let mut characteristic = mutex.lock();
`

I have temporarily fixed it for me by adding a check in the start of the function when the value is 0.

taks, 2024-11-18:
Thanks @vermasrijan19 

The event with ctxt.op as 0 is "Read characteristic event" ( `BLE_GATT_ACCESS_OP_READ_CHR`).
Is the connected device reading the characteristic?

taks, 2024-11-22:
I changed to use `try_lock()` in the `fix-daed` branch.
Could you check this?

vermasrijan19, 2024-12-11:
Sorry but i haven't had the bandwidth to be able to test this, I'll try and get back to you soon

vermasrijan19, 2024-12-27:
I have not been able to re-produce the issue but i think try_lock was a wrong suggestion by me as it would lead to read/other event failures if the characteristic is busy

vermasrijan19, 2024-12-27:
okay, so i have found a way to reliably reproduced the issue

I will update soon

vermasrijan19, 2024-12-27:
You are correct the issue seems to occur when a read/write operation happens during notify, this is not resolved on the `fix-deadlock` branch, I will try to provide more information

vermasrijan19, 2024-12-27:
Okay so , it looks like it is my bad, the deadlock was being caused due to a resource being locked before notify, it is not available 🤦.

But i have a couple of things , i would found a couple of things that I can submit PRs for

vermasrijan19, 2025-01-01:
Okay so follow up, there were more than 1 issues , it seems that if you keep notifying in a loop, without delay it does eventually cause a deadlock. Using try_lock in my loop on the deadlock branch seems to be the solution

PR #152 (2025-01-02): Fix deadlock in BLECharacteristic::handle_gap_event
Fix #151
closed 2025-01-02 by taks