taks/esp32-nimble #165: Can't compile if "CARGO_TARGET_DIR" is set

suitable-name, 2025-01-24:
I had the problem that even the most simple example wouldn't compile. Basically all the needed bindings to the esp-idf crates couldn't be found. It turned out, the problem is that I have the following in my (user) settings.json in vscode:
`"terminal.integrated.env.linux": {
        "CARGO_TARGET_DIR": "/mnt/ramdisk/${workspaceFolderBasename}/target"
    },`

Could you maybe check that the environment variable "CARGO_TARGET_DIR" gets handled correctly during build? Thanks!

taks, 2025-01-26:
If you set `CARGO_TARGET_DIR`, please add the following to `.cargo/config.toml`.
```toml
[env]
CARGO_WORKSPACE_DIR = { value = "", relative = true }

```

See [esp-idf-sys documentation](https://github.com/esp-rs/esp-idf-sys/blob/afed3054ba6ea1f985ced57a5d560a2c43db87f7/BUILD-OPTIONS.md#esp-idf-configuration)  for details.

suitable-name, 2025-01-27:
Thanks for the fast heads-up, I didn't see that before!

closed 2025-01-27 by suitable-name