taks/esp32-nimble #166: MTU 512 working for write_value but not for on_notify acting as Client

madhugoundla, 2025-01-31:
Hello,  

When I write a value for a characteristic while acting as a Client, I am able to write 512 bytes. However, on notification, I receive only 273 bytes. This also happened when acting as a Server, but the following setting resolved it, changing the received bytes from 273 to 512. This issue only persists on when acting as Client.

CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU=512

```
        let ble_device = BLEDevice::take();
        let mut ble_scan = BLEScan::new();
        block_on(async {
            let device = ble_scan
                .active_scan(true)
                .interval(100)
                .window(99)
                .start(ble_device, 10000, |device, data| {
                    for uuid in data.service_uuids() {
                        if uuid == ble_uuid {
                            ::log::info!("uuid value: {}", uuid);
                            return Some(*device);
                        }
                    }
                    None
                })
                .await
                .unwrap();

            if let Some(device) = device {
                let mut client = ble_device.new_client();
                client.on_connect(|client| {
                    client.update_conn_params(6, 12, 0, 1000).unwrap();
                });
                client
                    .connect(&device.addr())
                    .await
                    .expect("Failed to get service");

                let service = client
                    .get_service(ble_uuid)
                    .await
                    .expect("Failed to get service");
                ::log::info!("Service: {:?}", service);

                {
                    let uuid = uuid128!("00000000-a2d4-48ce-896b-4c76973373e6");
                    let characteristic = service
                        .get_characteristic(state_uuid)
                        .await
                        .expect("Failed to get service");
                    ::log::info!("characteristic: {}", characteristic);

                    let data = vec![0x01]; // Creates a vector containing 0x01
                    characteristic
                        .write_value(&data, false)
                        .await
                        .unwrap();

                    characteristic
                        .on_notify(|data| {
                            ::log::info!(
                                "characteristic: {}",
                                core::str::from_utf8(data).unwrap()
                            );
                        })
                        .subscribe_notify(true)
                        .await
                        .unwrap();
                }

                {
                    let recieve_uuid = uuid128!("00000000-a4dc-48ce-896b-4c76973373e6");
                    let receive_characteristic = service
                        .get_characteristic(recieve_uuid)
                        .await
                        .expect("Failed to get service");
                    ::log::info!(
                        "receive_characteristic: {}",
                        receive_characteristic
                    );
 
                    receive_characteristic
                        .on_notify(|data| {
                            let hex_string = hex::encode(data);

                            println!("Received chunk: {:?}", hex_string);
                            println!("Data length: {}", data.len());
                        
                        })
                        .subscribe_notify(false)
                        .await
                        .unwrap();
                }

                timer.delay(timer.tick_hz() * 1000).await.unwrap();
            }
        })
```

Am i missing anything? Why is 512 working for Write but not notify.

taks, 2025-02-03:
The problem is a bug in this library.

I fixed it in the [fix-long-data-on_notify](https://github.com/taks/esp32-nimble/tree/fix-long-data-on_notify) branch.
Could you check this?

madhugoundla, 2025-02-03:
Working Great, thanks for the FIx.

PR #167 (2025-02-03): Fix `on_notify` when data is long
Fix #166
closed 2025-02-03 by madhugoundla