taks/esp32-nimble #172: Timeout

Makuo12, 2025-02-16:
Sorry @taks , I thought this was for setting timeouts for connections

```rust
  /// * `timeout`: The timeout time in 10ms units before disconnecting.
  pub fn update_conn_params(
    &mut self,
    conn_handle: u16,
    min_interval: u16,
    max_interval: u16,
    latency: u16,
    timeout: u16,
```
However, the connection doesn't timeout at the right time that I set.

taks, 2025-02-17:
Could you explain what you have tried?

Makuo12, 2025-02-17:
```rust
    server.on_connect(move |server, desc| {
        ::log::info!("Client connected");
        // id = desc.conn_handle();
        server
            .update_conn_params(desc.conn_handle(), 24, 48, 0, 500)
            .unwrap();
        ::log::info!("Multi-connect support: start advertising");
        ble_advertising.lock().start().unwrap();
        let can_type = {
            let guard = match keypad_state.can_type.try_lock() {
                Ok(c) => *c,
                Err(_) => true
            };
            guard
        };
        if can_type {
            server.disconnect(desc.conn_handle()).unwrap();
            FreeRtos::delay_ms(1000);
        } else {
            connected_one.store(true, Ordering::Relaxed);
            conn_handle_one.store(desc.conn_handle(), Ordering::Relaxed);
        }
    });
```
I set it to 500 here `.update_conn_params(desc.conn_handle(), 24, 48, 0, 500)` which means it should timeout the connection in 5 seconds. Does the timeout also cause an automatic disconnect?

taks, 2025-02-18:
Yes, the timeout cause an automatic disconnect.

```rust
  server.on_disconnect(|_desc, reason| {
    ::log::info!("Client disconnected ({:?})", reason);
  });
```
With the above code, ```Client disconnected (Err(0x208)) ``` will be displayed at timeout.

How are you trying to cause the timeout?

taks, 2025-02-19:
Since BLE devices are constantly communicating, a timeout will not occur unless communication is lost. (e.g., by the device moving away).

Makuo12, 2025-02-24:
Thanks @taks, I understand now.

closed 2025-02-24 by Makuo12