taks/esp32-nimble #175: Cargo check fails with 0.10.2

kirillolenev-dm, 2025-02-20:
```
error[E0432]: unresolved import `esp_idf_sys::ble_gap_conn_desc`
 --> /Users/kirill/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.10.2/src/server/ble_conn_desc.rs:3:5
  |
3 | use esp_idf_sys::ble_gap_conn_desc;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `ble_gap_conn_desc` in `sys`

error[E0432]: unresolved imports `esp_idf_sys::ble_uuid_any_t`, `esp_idf_sys::ble_uuid_cmp`
 --> /Users/kirill/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.10.2/src/server/ble_descriptor.rs:6:19
  |
6 | use esp_idf_sys::{ble_uuid_any_t, ble_uuid_cmp};
  |                   ^^^^^^^^^^^^^^  ^^^^^^^^^^^^ no `ble_uuid_cmp` in `sys`
  |                   |
  |                   no `ble_uuid_any_t` in `sys`

```
And many many other similar errors.
**cargo build**/compilation and everything is working ok.

```
[package]
name = "esp-idf-shashlik"
version = "0.1.0"
authors = ["expl <kpetrikas@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "esp-idf-shashlik"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = 's'
overflow-checks = false
codegen-units = 1
lto = 'fat'

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = []

experimental = ["esp-idf-svc/experimental"]

[dependencies]
log = "0.4"
esp-idf-svc = { version = "0.51", features = ["alloc", "experimental"]}
embedded-hal = "1"
itybity = { version = "0.2", default-features = false }
nalgebra = { version = "0.33.2"}
micromath = "2.1.0"
bresenham = "0.1.1"
byte-slice-cast = "1.2.2"
enumset = "1.1.5"
itertools = "0.14.0"
flatbuffers = "25.2.10"
postcard = "1.1.1"
serde = "1.0"
googleprojection = "1.2"
esp32-nimble = "0.10.2"

[build-dependencies]
embuild = "0.33"

[package.metadata.esp-idf-sys]
esp_idf_tools_install_dir = "global"
esp_idf_sdkconfig = "sdkconfig"
esp_idf_sdkconfig_defaults = "sdkconfig.defaults"
esp_idf_component_manager = "y"
extra_components = [
    { bindings_header = "include/bindings.h" }
]
```

```
[build]
rustflags = [
  "-C", "link-arg=-nostartfiles",
  "--cfg", "esp_idf_bt_enabled",
  "--cfg", "esp_idf_bt_ble_enabled",
  "--cfg", "espidf_time64"
]

target = "xtensa-esp32s3-espidf"

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor"
rustflags = [ "--cfg", "espidf_time64"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32s3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.4"
ESP_LOG="INFO"
```

```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000

# Use this to set FreeRTOS kernel tick frequency to 1000 Hz (100 Hz by default).
# This allows to use 1 ms granularity for thread sleeps (10 ms by default).
#CONFIG_FREERTOS_HZ=1000

# Workaround for https://github.com/espressif/esp-idf/issues/7631
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=n
CONFIG_SOC_LCD_I80_SUPPORTED=y
CONFIG_SOC_LCD_RGB_SUPPORTED=y
CONFIG_SOC_LCD_I80_BUSES=1
CONFIG_SOC_LCD_RGB_PANELS=1
CONFIG_SOC_LCD_I80_BUS_WIDTH=16
CONFIG_SOC_LCD_RGB_DATA_WIDTH=16
CONFIG_SOC_LCD_RGB_SUPPORTED=y
CONFIG_SOC_LCD_SUPPORT_RGB_YUV_CONV=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_ESP32S3_SPIRAM_SUPPORT=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_TASK_WDT=n
CONFIG_ESP_TASK_WDT=n
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_SPIRAM_XIP_FROM_PSRAM=y
CONFIG_ESP32S3_DATA_CACHE_32KB=n
CONFIG_ESP32S3_DATA_CACHE_64KB=y

CONFIG_BT_ENABLED=y
CONFIG_BT_BLE_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y
```

taks, 2025-02-24:
- Do you mean that `cargo build` succeeds but `cargo check` fails?
- Please try deleting `esp_idf_tools_install_dir = "global"` and see if the errors still occur.

kirillolenev-dm, 2025-02-24:
@taks 
1. Exactly, `cargo build` succeeds.
2. Tried, didn't help, the same errors.

taks, 2025-02-25:
Could you please run `cargo clean && cargo check -v 2> check_log.txt` and `cargo clean && cargo build -v 2> build_log.txt` and attach the logs?

kirillolenev-dm, 2025-02-25:
@taks Some new finding here actually. The problem isn't with `cargo check` but with the build type.
`cargo build` fails but `cargo build --release` is ok. I attached logs for both of them.

[build_log_release.txt](https://github.com/user-attachments/files/18958946/build_log_release.txt)
[build_log_debug.txt](https://github.com/user-attachments/files/18958945/build_log_debug.txt)

taks, 2025-02-26:
It seems that the sdkconfig(.default) isn't loaded in the debug build.

What is the file name below?
```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000
...
```

kirillolenev-dm, 2025-02-26:
@taks you're right! The name of the file was incorrect. Now everything works great. Thanks for your help. Closing it.

closed 2025-02-26 by kirillolenev-dm