taks/esp32-nimble #178: Unable to scan Bluetooth earphones

silent-rain, 2025-03-16:
Unable to scan Bluetooth earphones and phone Bluetooth, unable to display corresponding Bluetooth names.

Could you please provide some guidance? Thank you

## device
[env]
MCU = "esp32s3"
ESP_IDF_VERSION = "v5.3.2"

phone name : sr
earphones name: bt

## Example of operation
cargo run -r --example ble_client

```rust
    let device = ble_scan
      .active_scan(true)
      .interval(100)
      .window(99)
      .start(ble_device, 10000, |device, data| {
        log::info!("Advertised Device: ({:?}, {:?}", device, data);
        log::info!(
          "Advertised Device name: ({:?}, {:?})",
          data.name(),
          device.adv_type(),
        );
        log::info!("=======================");

        if let Some(name) = data.name() {
          if name.contains_str("ESP32") {
            return Some(*device);
          }
        }
        None
      })
      .await?;
```

## Log output
```text
I (666) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 0D:D5:00:71:E6:A9(random), adv_type: NonconnInd, rssi: -86, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 6, payload: [1, 9, 32, 34, 82, 218, 251, 206, 151, 250, 62, 194, 199, 241, 156, 218, 224, 79, 87, 238, 242, 202, 18, 156, 5, 185, 11] } }
I (696) ble_client: Advertised Device name: (None, NonconnInd)
I (696) ble_client: =======================
I (706) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 33:E5:A4:F9:D6:0E(random), adv_type: NonconnInd, rssi: -88, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 6, payload: [1, 9, 32, 2, 188, 170, 17, 222, 191, 144, 86, 202, 61, 151, 248, 244, 17, 106, 10, 18, 140, 176, 110, 153, 163, 193, 96] } }
I (736) ble_client: Advertised Device name: (None, NonconnInd)
I (746) ble_client: =======================
I (756) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 3E:01:80:47:2E:DB(random), adv_type: NonconnInd, rssi: -90, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 911, payload: [18, 17, 0, 17, 2, 99, 230, 157, 142, 229, 179, 176, 36, 0, 0, 56, 192, 14, 4] } }
I (786) ble_client: Advertised Device name: (None, NonconnInd)
I (786) ble_client: =======================
I (796) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 12:17:9E:3A:EF:BB(random), adv_type: NonconnInd, rssi: -84, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 6, payload: [1, 9, 32, 34, 32, 83, 16, 99, 107, 132, 62, 60, 55, 50, 251, 110, 181, 96, 24, 250, 35, 195, 112, 248, 179, 164, 115] } }
I (826) ble_client: Advertised Device name: (None, NonconnInd)
I (836) ble_client: =======================
I (846) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 4F:34:C8:A2:CA:4D(random), adv_type: NonconnInd, rssi: -87, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { adv_flags: AdvFlag(DiscGeneral | SimultaneousController | SimultaneousHost), manufacture_data: ManufactureData { company_identifier: 76, payload: [22, 8, 0, 214, 138, 57, 8, 36, 120, 17] } }
I (876) ble_client: Advertised Device name: (None, NonconnInd)
I (886) ble_client: =======================
I (1046) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 65:96:47:14:26:A3(random), adv_type: Ind, rssi: -64, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { adv_flags: AdvFlag(DiscGeneral | SimultaneousController | SimultaneousHost), tx_power: 12, manufacture_data: ManufactureData { company_identifier: 76, payload: [16, 5, 3, 28, 218, 8, 38] } }
I (1066) ble_client: Advertised Device name: (None, Ind)
I (1076) ble_client: =======================
I (1076) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 65:96:47:14:26:A3(random), adv_type: ScanResponse, rssi: -64, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData
I (1096) ble_client: Advertised Device name: (None, ScanResponse)
I (1106) ble_client: =======================
I (1106) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 34:65:92:70:3F:DF(random), adv_type: NonconnInd, rssi: -83, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 6, payload: [1, 9, 33, 10, 163, 117, 199, 73, 38, 24, 77, 114, 115, 105, 99, 107] } }
I (1136) ble_client: Advertised Device name: (None, NonconnInd)
I (1146) ble_client: =======================
I (1676) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: C9:98:1C:36:B5:6C(random), adv_type: NonconnInd, rssi: -64, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 76, payload: [18, 2, 0, 0] } }
I (1686) ble_client: Advertised Device name: (None, NonconnInd)
I (1696) ble_client: =======================
I (1836) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: D6:24:6F:43:F7:B8, adv_type: Ind, rssi: -95, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { adv_flags: AdvFlag(DiscLimited | BrEdrUnsupported), manufacture_data: ManufactureData { company_identifier: 27833, payload: [214, 36, 111, 67, 247, 184] }, service_uuids: [0xfee7, 0xfba0] }
I (1856) ble_client: Advertised Device name: (None, Ind)
I (1866) ble_client: =======================
I (1866) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: D6:24:6F:43:F7:B8, adv_type: ScanResponse, rssi: -91, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { name: "KEY" }
I (1886) ble_client: Advertised Device name: (Some("KEY"), ScanResponse)
I (1896) ble_client: =======================
I (2386) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: C4:10:F2:A5:F7:0B(random), adv_type: NonconnInd, rssi: -87, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { manufacture_data: ManufactureData { company_identifier: 76, payload: [18, 2, 0, 1] } }
I (2396) ble_client: Advertised Device name: (None, NonconnInd)
I (2406) ble_client: =======================
I (7436) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 84:46:93:68:0B:9A, adv_type: Ind, rssi: -90, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { adv_flags: AdvFlag(DiscGeneral | BrEdrUnsupported), service_data: BLEServiceData { uuid: 0xfe95, service_data: [16, 89, 217, 48, 164, 154, 11, 104, 147, 70, 132] } }
I (7456) ble_client: Advertised Device name: (None, Ind)
I (7466) ble_client: =======================
I (7476) ble_client: Advertised Device: (BLEAdvertisedDevice { addr: 84:46:93:68:0B:9A, adv_type: ScanResponse, rssi: -88, sid: 255, prim_phy: Phy1M, sec_phy: None, periodic_itvl: 0 }, BLEAdvertisedData { name: "Mijia Scale S400 0B9A" }
I (7496) ble_client: Advertised Device name: (Some("Mijia Scale S400 0B9A"), ScanResponse)
I (7506) ble_client: =======================
I (10616) main_task: Returned from app_main()
```

taks, 2025-03-17:
- Are your earphones Bluetooth or Bluetooth Low Energy?
- Are you running the BLE program on your phone?
  You cannot scan from esp32 without running the BLE peripheral program.

silent-rain, 2025-03-18:
This is a low-power Bluetooth earphone

silent-rain, 2025-03-31:
After confirming, the headphones are indeed not low-power. Sorry to bother you

closed 2025-03-31 by silent-rain